foldynl/QLog #590: RcvNr for contest doesn't work

1Codealot, 2025-02-05:
Used this when participating in a contest yesterday, went to transcribe log today, RcvNr just doesn't exist.

foldynl, 2025-02-05:
Were all the conditions described on the [WIKI](https://github.com/foldynl/QLog/wiki/Contest-Support#important-notes) met?

1Codealot, 2025-02-05:
No because my contest isn't part of the contest ID.

What reason is there for it to not save it if it isn't in "contest mode"? this is really annoying.

foldynl, 2025-02-05:
Each contest has a Contest ID. Even if it is not present in ADIF Contest ID, each contest should have its ID assigned by the organizers.

>>> What reason is there for it to not save it if it isn't in "contest mode"? this is really annoying.

I apologize for the problems, but logging a contest without trying it first is not good practice. 

I can answer your question that everything is on WIKI. It is necessary for the contest to start somehow . For QLog, the starting point is the first QSO with Contest ID. A contest is not just RcvNr, it can be anything else (Age, Power, ITU, CQ etc). So the only way to clearly recognize that it is a contest is to fill in the Contest ID. 

If you know any general conditions how to identify the first QSO in a general contest, it is welcome.

foldynl, 2025-02-06:
Sorry to bring this up again, but how could you write RcvNr when this field is inactive if the contest ID is not filled in?

1Codealot, 2025-02-06:
By having a custom activity and adding RcvNr into one of the rows

[image]

Now I know about the contestID, i just didn't rtfm :-/

commit 3b7aa1e6 (2025-02-06): Fixed #590 - Suppressed the ability to edit Contest fields after the start
closed 2025-02-14 by foldynl