foldynl/QLog #591: Stall and crash with Windows 11

iu0dhv, 2025-02-06:
I have experience several stall of the log with subsequent crash or need to restart. 
I have now identified a specific case when this problem arise. If I double click on a qso in the log, the windows with all the details opens, and then the software remains unresponsive, with the need to restart. 
It is practically in possible to open a specific QSO in the log to change something, as the program becomes unusable. 
Can someone help to fix this problem?
74, Peter

foldynl, 2025-02-07:
Does it crash only for specific QSO or is it for random QSOs from your log?

iu0dhv, 2025-02-07:
Hello the log stalls very very frequently and each time I try to edit a
QSO. So not on specific ones but not even randomly. It does so in all
occurrences I have tried.
Moreover, yesterday I was trying the very convenient command to massive
updates for past QSOs. However, even that function did not work as
expected. Selecting bundles of QSOs, only on very few occasions the updates
went through flawlessly. Most of the times, however, the bar showing the
percentage of completion stopped at a percentage that, depending on the
size of the QSO selection, corresponded always to 1 QSO. So I had to stop,
because I could not go on with the massive updates.
I have also tried to close the only other application that was open, i.e.
JTDX, but nothing changed.
To give you some more context, I have a fairly new and performing laptop
and no issues with any other and much heavier logs (e.g. Log4om).
I hope the issue can be solved as I like the log very much :-)
73
Peter, IU0DHV

PS I also have troubles in connecting the radio (FT DX10) via Cat, but for
the time being this is a less important issue.

foldynl, 2025-02-07:
can you send me a debug log? Following instructions below:

1) `cmd`
2) `cd` to the directory where QLOG is installed
3) `set QT_LOGGING_RULES=qlog.*=true`
4) `qlog.exe -d`
5) double click on some QSO
6) send me a log file `C:/Users/<USER>/AppData/Local/hamradio/QLog/qlog_debug_yyyyMMddhhmmss.log` (it will probably be quite big so zip it).

iu0dhv, 2025-02-07:
Hello again,
please find attached the debug log zipped, obtained following your
instructions.
Please let me know if the issue can be resolved.
Best
Peter, IU0DHV

aa5sh, 2025-02-07:
@iu0dhv  it doesn't look like the attachment made it.  You may need to go the github website and attach it there.  If you reply to an email it doesn't bring the attachments in.

Michael, AA5SH

iu0dhv, 2025-02-07:
> Maybe it's too big. send me a link to my email where I can download it;
> google drive or another online service

It is just 180kb Zip file....

iu0dhv, 2025-02-07:
I have now shared the link to the file

iu0dhv, 2025-02-07:
[qlog_debug_20250207210603.zip](https://github.com/user-attachments/files/18713561/qlog_debug_20250207210603.zip)

iu0dhv, 2025-02-07:
I have now shared the zip file direcly here on github

foldynl, 2025-02-08:
That's strange. Unfortunately, I don't see any reason why the dialog would be blocked.

The strange thing is that I can see the last event updating the time in the clock widget, which is not related to QSODetail. The application is somehow still alive because I can see CW thread activity, but the main thread is not working. Unfortunately, I am not able to identify the cause, even with such a detailed list.

I have a few questions:

* Did you see the map in the QSO Dialog, or did it not appear?
* Would it be possible to send a video showing how QLog is freezing?
* Does the application crash on its own, or do you terminate it through the OS?

iu0dhv, 2025-02-08:
Hello I am sorry that there is not obvious solution to the problem. I am attaching here a short video that shows that kind of problem that I am having. The situation shown in the video occurs any time I try to make any change to a QSO. 
As per your questions, as you will see in the video.
I have the maps open
I don't have the clock video open
I have to crash the log myself. IT DOES NOT crash on its own, but it stops working. 
Take a look at the video

[WhatsApp Video 2025-02-08 at 14.29.23.zip](https://github.com/user-attachments/files/18718982/WhatsApp.Video.2025-02-08.at.14.29.23.zip)

foldynl, 2025-02-08:
Thanks for the video. Unfortunately, I do know how to help you. If you're reporting issues in different places, it could be due to a corrupted database. However, this only happens in exceptional cases - OS Crash, Power failure, Memory issue, Disk issue etc.

The only thing you could try is running QLog with the `-n test` switch. This is used during development when I need to create another instance of QLog with a new database. It won't affect your current database but will create a separate instance of QLog (new database, new configuration). It means that you'll need to go through profile setup again.

After that, try to insert a QSO and performing the operation that caused QLog to freeze.

iu0dhv, 2025-02-08:
Could it be that something on my log file (ADI) causes the freeze given that with thre fake QSo everything seems to be ok?

foldynl, 2025-02-08:
If you uninstall and reinstall QLog, your data will not be deleted. Only the application and libraries are removed, but the QSO database, passwords, and settings remain.

If your test instance worked, the issue might be a corrupted database file. Ideally, export all QSOs if possible. If it is not possible, check `C:/Users/<USER>/AppData/Local/hamradio/QLog` for any ADX backups. If they exist, back them up somewhere. Then, delete the entire `C:/Users/<USER>/AppData/Local/hamradio/QLog` folder.

After that, you will also need to delete the entire registry entry at:
`HKEY_CURRENT_USER\Software\Hamradio\QLog`

Once done, start QLog again, and it will guide you through the initial setup. Finally, import the ADX backup and check if everything works correctly.

>>> Could it be that something on my log file (ADI) causes the freeze given that with thre fake QSo everything seems to be ok?

I don't know if I don't have this file.

iu0dhv, 2025-02-08:
I have tried deleteing evrything. I have imported again a brand new adif with basic fields (not too many). The import went fine. I was going to set the connection to QRZ, then I changed my mind and went back to try to edit a new qso before connecting anything, and gues what? freeze! :-(
Do not what to say. I could share with you the adif, but it a completely standard adi file with not fancy info in it

iu0dhv, 2025-02-08:
[image]
this is what I have....

foldynl, 2025-02-09:
I don't think it's a problem in ADI but we can try. Please send it to my private email - my callsign @ gmail.com

iu0dhv, 2025-02-09:
I don't think either is a ADI problem. What I don't understand is why the
problem occurs only with Qlog and not with other logs.

foldynl, 2025-02-09:
There's nothing unusual about it since QLog is a different application.

To be honest, I'm sorry but I'm not able to help you. I don't know where the issue is. It could also be an issue of QT library. The next release will come out with an updated QT library, maybe that will help you. I don't know.

iu0dhv, 2025-02-09:
Thank you for the support. I will try with a future version of QLog,
hopefully it will work for me.
73 IU0DHV

iu0dhv, 2025-03-22:
Hello I am now using the latest version (0.42.2) and the freezing problem has neve occured! So the issue, whatever it was, has been solved. Thanks

foldynl, 2025-03-23:
Thanks for the report. It's a shame I don't know what caused it, because when I go through the Changelog I don't see anything that could improve it. But the important thing is that it works for you now.

closed 2025-03-23 by foldynl