foldynl/QLog #595: Display my callsign as a column

aronszabo, 2025-02-09:
There seems to be no way to display my own callsign in the log table. If I right click the column headers, "Owner callsign" appears two times, but both of them display an empty column.

aa5sh, 2025-02-10:
For me it is stored in the field 'Logging Station Callsign' under Others:

[image]

foldynl, 2025-02-10:
I think it is not very clear in the hamradio world. So I add from the ADIF spec:

**STATION_CALLSIGN** - the logging station's callsign (the callsign used over the air) . If STATION_CALLSIGN is absent, [OPERATOR](https://adif.org/315/ADIF_315.htm#QSO_Field_OPERATOR) shall be treated as both the logging station's callsign and the logging operator's callsign

**OWNER_CALLSIGN** - the callsign of the owner of the station used to log the contact (the
callsign of the [OPERATOR's](https://adif.org/315/ADIF_315.htm#QSO_Field_OPERATOR) host). if [OWNER_CALLSIGN](https://adif.org/315/ADIF_315.htm#QSO_Field_OWNER_CALLSIGN) is absent, [STATION_CALLSIGN](https://adif.org/315/ADIF_315.htm#QSO_Field_STATION_CALLSIGN) shall be treated as both the [logging station's](https://adif.org/315/ADIF_315.htm#QSO_Field_STATION_CALLSIGN) callsign and the callsign of the owner of the station

**OPERATOR** - the logging operator's callsign. the logging operator's callsign. If [STATION_CALLSIGN](https://adif.org/315/ADIF_315.htm#QSO_Field_STATION_CALLSIGN) is absent, [OPERATOR](https://adif.org/315/ADIF_315.htm#QSO_Field_OPERATOR) shall be treated as both the [logging station's](https://adif.org/315/ADIF_315.htm#QSO_Field_STATION_CALLSIGN) callsign and the logging operator's callsign

aronszabo, 2025-02-14:
Thank you, I did not know about this "Column visibility setting" dialog window, I only found the context menu when right clicking the headers. There seem to be two different sets of available columns.

aa5sh, 2025-02-14:
They should be the same list just that in the column visibility form they are grouped and easier to find.  You can see it is here as well but had to hunt for it a bit more:

[image]

[image]