foldynl/QLog #600: Time on Info of DX contact

dimarmnes, 2025-02-16:
I have noticed in version 0.42.0 that the time in the Contact Info section has changed its format, not being able to display it completely.

[image]
Before

[image]
v 0.42.0

foldynl, 2025-02-16:
What OS are you using?

dimarmnes, 2025-02-16:
Sorry, Manjaro Gnome, QLog Flatpak version
Diego Carratini

foldynl, 2025-02-16:
please, could you send me a result of following command ?

```
env | grep LC_
```

and screenshot of QLog->Help->About ?

dimarmnes, 2025-02-16:
Follow,

LC_IDENTIFICATION=es_AR.UTF-8
LC_MEASUREMENT=es_AR.UTF-8
LC_NUMERIC=es_AR.UTF-8
LC_TIME=es_AR.UTF-8
LC_PAPER=es_AR.UTF-8
LC_MONETARY=es_AR.UTF-8
LC_TELEPHONE=es_AR.UTF-8
LC_NAME=es_AR.UTF-8
LC_ADDRESS=es_AR.UTF-8

[image]

commit 0fb78e0b (2025-02-22): Fixed #600: Unexpected Timezone info
closed 2025-02-22 by foldynl