foldynl/QLog #604: program closes by itself

TecnoTalarn, 2025-02-21:
Hello, since the last update I'm seeing that the program closes by itself from time to time. Until version 0.41 it had never happened.
I'm using the flatpak version on Ubuntu.

root@radio:/var/log#  cat syslog | grep qlog | grep segfault

2025-02-20T06:01:41.194651+01:00 radio kernel: qlog[3802]: segfault at 0 ip 0000000000000000 sp 00007ffdad571788 error 14 in qlog[5a3701ff3000+76000] likely on CPU 0 (core 0, socket 0)

2025-02-20T07:55:57.193550+01:00 radio kernel: qlog[14860]: segfault at 612b72d585df ip 0000612d2f47a6d6 sp 00007fff5a677e50 error 4 in qlog[612d2f36c000+442000] likely on CPU 3 (core 1, socket 0)

2025-02-20T07:57:06.193439+01:00 radio kernel: qlog[15338]: segfault at f1 ip 000059576dbe76d6 sp 00007fff4286b770 error 4 in qlog[59576dad9000+442000] likely on CPU 1 (core 1, socket 0)

2025-02-20T11:49:18.194944+01:00 radio kernel: qlog[15480]: segfault at 62ea504a5bb0 ip 000062ea504a5bb0 sp 00007fffe8b29a38 error 15 likely on CPU 1 (core 1, socket 0)

2025-02-20T15:22:49.194439+01:00 radio kernel: qlog[27189]: segfault at 5ccc878dcd85 ip 00005cc93dae06d6 sp 00007ffd612b12f0 error 4 in qlog[5cc93d9d2000+442000] likely on CPU 1 (core 1, socket 0)

2025-02-21T11:23:38.195456+01:00 radio kernel: qlog[45150]: segfault at 0 ip 0000000000000000 sp 00007ffc58a76fa8 error 14 in qlog[6252f3b57000+76000] likely on CPU 1 (core 1, socket 0)

root@radio:/var/log#

foldynl, 2025-02-21:
Yes. Unfortunately, there is an issue when QLog is NOT connected to the internet or there is an issue with connection to DXC Stat server or wakes up from sleep mode then QLog crashes. I am working on a fix now.

foldynl, 2025-02-21:
The fix will be available very soon. I probably won't have time to do it today, but I'll try to get it ready tomorrow.

foldynl, 2025-02-22:
The fix is available - v0.42.1

closed 2025-02-22 by foldynl