foldynl/QLog #607: Problems with connection with 991A and CW

hb9fbg, 2025-02-24:
Hi, I saw that a friend uses QLog with satisfaction and I downloaded it for Windows. 
Everything works ok and connects to my Yaesu 991a but I can't configure the connection with the CW.
This configuration is intended for my portable activities and previously I used another software (less beautiful and performing) and the CW macro dialogue towards 991 passed through the USB port.
I did several tests but when I arrive to configure the USB port it does not appear (no port at all) but I would like to use the PC connection to 991
My portable PC (recovered from my son's "old" gaming PC) is Windows 11
Can I ask where and what I'm doing wrong?
Thanks in advance for any help
73s Mauro HB9FBG

PS: the log (50k QSOs) is managed very well and is very beautiful... congratulations!

[image]

foldynl, 2025-02-27:
Hello, If you want to send CW over CAT then it is needed to define CW key as Mode = `Morse over CAT`

[image]

And then assign the key to the Rig

[image]

I am not sure whether 991 can send a morse over CAT correctly - it depends on Hamlib implementation.

foldynl, 2025-03-04:
Unfortunately, I have never used HRD. 

I quickly went through the 991A manual, and I’m afraid I have bad news. QLog cannot provide full support for CW as you would like. QLog does not support keying via a COM port (COM4). QLog can only send text to keyers such as Winkey, FLDigi, CWDaemon, or directly to the rig via CAT. The last-mentioned function is implemented for the 991A, but unfortunately, only with limited capabilities—991A only accepts predefined CW macro numbers via CAT commands. This means that with QLog, you would only be able to send the numbers of these macros.

hb9fbg, 2025-03-04:
TNX!

vazquezrodjoel, 2025-04-05:
I have a 991a, and I send CW with CWDaemon, controlled with Hamlib, but on Linux.  It works perfect.  The only think I use different is that I have another little computer just running CWDaemon and rigctld constantly, separate from the computer I run qlog.  This is so I can use it with other computers over the network if I want to roam around the house (in that case I don't use qlog).  But works perfect.  Using it on the same computer would make it easy.  With only one USB cable, you should see 2 ports, assign one to CW the other to QLog.  If you are not afraid of, you could try similar approach with like a raspberry pi connected to your radio, then point Qlog to the RPI...