foldynl/QLog #612: Help with FT dx10 connection but also a good news

iu0dhv, 2025-03-01:
Hello I have downloaded the latest version of QLog and I very happy that it seems to work smoothly. After two updates I am not experiencing anymore the fatal freezings of the log that I used to have with with the t-2 version and that forced me to revert to the another log. So this is a very good news.

Now, what I would like to do, if possible, is to: connect the log to the radio, run JTdx at the same time, and have the cw console working at the same time. Has anyone been able to do it? 

I have not at the moment.

With hamlib the radio does not connect at all (strange). Does the log starts hamlib by itself? How could I have the radio connected with hamlib?

With Omnirig, it connected (and JTDX run), but than after several attemps it does not coonnect anymore :-(. And also, should it be possible to have the cw working with Omnirig or only with hamlib?

Once sorted this connection issue  in a stable way, and having JTDX working, I would like to have also the CW console working.

Finally, it would nice to a clear indication of wether the log is connected or not with the radio (like a green led...)

 
73

Peter, IU0DHV

aa5sh, 2025-03-01:
I have a Yaesu FTDX-101D (I did just buy a FTDX-10 but haven't had a chance to connect it up at my remote site) which is very similar I would imagine with CAT control.  What I do which works well for me is launch rigctld (which is part of HamLib) in a terminal (similar to command prompt in windows) before I open QLog or WSJT-x.  Doing it this way QLog, WSJTX and a custom app I have to control my Steppir and SPE amp can all talk to my radio nicely.

I connect to my radio with rigctld with the following command:
`rigctld -m 1044 -s 38400 -r /dev/cu.SLAB_USBtoUART5`

On windows for the FTDX-10 it would be something like this:
`rigctld -m 1042 -s 38400 -r COM5 `

You would need to update the baud rate and com port appropriately.  Then in QLog here are my settings which should be similar to windows setup:

[image]

and in WSJTX

[image]

For the connection to the rig you may want to open the Rig Control window in QLog it will show red or green:

[image]

iu0dhv, 2025-03-01:
Hello lunching rigctld as per your suggestion in the second line (but change the port to Com4 instead of 5) and setting the radio in Qlog and in Jtdx as suggested makes both working (at least if seems so at the moment). Thanks!

To make the solution more effective though it would be good to have rigctld lauchned automatically at the opening of QLog, if instructed to do some in the settings, instead of having to type che ommand lines by hand each time.

That said, has anyone have any indication on how to make the cw over cat working?

73

aa5sh, 2025-03-01:
if you create a cat port like this then select like in screenshot above it should work: 

[image]

aa5sh, 2025-03-01:
noticed the assigned cw keyed option on the rig settings in qlog.

iu0dhv, 2025-03-02:
Hello the profile has been created and once I open the CW window in the log the name of the profile is green so I assume it works fine. But it doesn't transmit. So maybe I have to play with the settings of the radio, even though with other logs (e.g. contest logs) it woks.

iu0dhv, 2025-03-02:
Actually just to confirm that the cw connection works, if I change the cw from the log it does change it to the keyer of radio. How it does not transmit

aa5sh, 2025-03-02:
On my 101 I need to make sure the bk-in button is pressed.  Basically need to have the radio setup so it is good to go with my cw key plugged into the radio and it works great with QLog.

iu0dhv, 2025-03-02:
I do have the Bk-in on and the keyer buttons on.  Do the macro from Qlog work? in my case the don't. But for instance they work in QARtest and N1MM.
Any other ideas or setting you can share?

iu0dhv, 2025-03-02:
yes with both QARtest and N1MM I am using the macros and the keyboard to transmit in cw

aa5sh, 2025-03-02:
But how is QARTest and N1MM generating the CW.  Using a serial port and RTS/DTR or using the CAT commands of the radio?  

For QLog you would need to use either CAT, Winkey or similar.  QLog does not generate the CW using RTS/DTR like N1MM does.

iu0dhv, 2025-03-02:
Actually QARTest (and N1MM) generates CW and uses the COM port and RST/DTR. 
So if I understand correctly  QLog does not generate CW but requires an external keyer, being it the one of the radio or another keyer,
So, what does the CAT selection do? Should it allow to use the keyer of the radio?

aa5sh, 2025-03-02:
Yes many of the modern rigs support CAT generation over normal rig control.  That is what I use with my radios with the settings I shared above.

iu0dhv, 2025-03-02:
thanks. Any further ideas to make the FT dX 10 sending CW over cat with Qlog?

aa5sh, 2025-03-03:
Can you send a screenshot of your CW Keyer and Rig settings pages from QLog showing the details of what you have configured?

iu0dhv, 2025-03-03:
[image]

[image]
[image]
[image]
[image]

iu0dhv, 2025-03-03:
just shared the settings of QLog and of the FT dx 10 if the can be of help

aa5sh, 2025-03-03:
Those all look ok.  One last screenshot can you share your Radio settings panel in QLog?

iu0dhv, 2025-03-03:
Here it is 
[image]

aa5sh, 2025-03-03:
Ok you need to select your 'cw cat' cw keyere here:

[image]

and it should work.

iu0dhv, 2025-03-03:
How do I find it?

iu0dhv, 2025-03-03:
I hoped so but I tried and nothing happened. :-(

aa5sh, 2025-03-03:
Ok the only thing I can think to validate is that your rig was in CW mode with Bk-In enabled?  I will likely be going to my remote site the next couple days to replace some coax and I can test it when I get there.  But those settings I would imagine should have worked.

iu0dhv, 2025-03-03:
Yes CW mode with Bk-in enabled...

iu0dhv, 2025-03-03:
Hello I have discovered why the radio did not transmit. I was using VFO-B (as in VFO I was in data mode). When using VFO-A the radio started transmitting. 

However when I type the text everything seems to be ok. If I use a macro after a few letters it stop and the radio gets disconnected with the following error message

[image]

why does it get timed out?