foldynl/QLog #615: Close/Maximize/Minimize button disappear when rotor window is activated after restart

DL1SP, 2025-03-08:
Hello Guys, 

many thanks for this great software.
Two years ago i change from windows 10 to Linux Mint 21.3 Cinnamon. 
I notices today that my Close/Maximize/Minimize button disappear when rotor window is activated and i restart the software. 
after some try's i figured that when i delete the .conf file at "$HOME/.config/hamradio/QLog.conf" and restart the software (without any change) the lose/Maximize/Minimize button are still there but as soon as i add the rotor window and after the next restart of the software the lose/Maximize/Minimize button are gone. Even if i remove the rotor windows again the button are still gone. 

maybe some can help me whats wrong on my system. 

Thanks. 

73 de Simon (DL1SP)
greatings from Germany.

foldynl, 2025-03-09:
Hi Simon,

unfortunately, it is know issue under Linux. There is an issue in QT Library (GUI library). 

I'm trying to fix it somehow long time but if I fix it, another issue in the Qt Library appears. It is not completely related to the Rotator Widget. 

I don't recommend deleting the config file. It's better to delete only these two parameters: `General`→`geometry` and `General`→`windowState`. If you are using the Activity Manage then it is good to clear the `Window Arrangement` for the particular profile.

foldynl, 2025-03-09:
Please, could you run this command in the terminal (command line) and send the result?

`ps -ef | grep wayl`

DL1SP, 2025-03-09:
Hello Ladislav, 

Many thanks for your fast answer. 
coding is like fighting with the enemy from all sides;-)

the output from  "ps -ef | grep wayl" is:

simon       2545    2529  0 18:19 pts/0    00:00:00 grep --color=auto wayl

but today the button are there with the rotator windows activated (and WSJT-X window) . 

73 de Simon DL1SP

DL1SP, 2025-03-22:
Hello Ladislav,

two weeks later and the annoying bug still here. 

73 de Simon DL1SP

foldynl, 2025-03-22:
Yes, you are right. A fix is planned for the next release. Keep in mind that I am fixing something that is not under my control, so I am not sure if it will help. 

BTW, I determine the order of the fixes if no one implements the fix. Also, I don't think this is a critical bug.

foldynl, 2025-04-04:
should be fixed in 0.43...please, confirm

DL1SP, 2025-05-28:
Hello Ladislav,

yes working very well. Thank you very much. Even if it (is) was not your fault. 

for the records you need to delete the eneral→geometry and General→windowState in the conf file. 

73 from Simon DL1SP

closed 2025-05-30 by foldynl