foldynl/QLog #617: Online Map context menu lost

dl2ki, 2025-03-15:
Hi there,

I noticed that the online map no longer has a context menu. Only this image is displayed. The cursor is displayed as a "moving hand" in this area. The online map also does not show a path to the station that is entered in the New QSO window.

[image]

dl2ki, 2025-03-15:
Just started QLog again. Now the online map works. Is it possible that this is due to a high utilization of the DX cluster?
The menu for layer selection is also displayed again.

[image]

Issue can then be closed in my view after reading.

foldynl, 2025-03-15:
Maybe there was an internet connection issue—online map use many online resources, and if any of them is missing, it does not work correctly.

closed 2025-03-15 by foldynl