foldynl/QLog #618: Speed pot doesn't seem to work with a WinKeyer

mi0hoz, 2025-03-15:
When I change the cw speed on my WinKeyer with the speed pot it shows on the screen correctly but the actual speed of the sent morse doesn't change. If you change the speed on the console it works perfectly.

dl2ki, 2025-03-15:
I cannot confirm this with my K3NG-Winkeyer. Speed changes with the potentiometer are applied correctly.

aa5sh, 2025-03-21:
I was playing with this some last night withy my K3NG keyer (Funtronics FK-11) I bought a while back.  It does play a little weird.  The speed pot does seem to change the keyer speed for when I use my paddle, but it doesn't change the speed that the CW is sent when I use a macro button in QLog at times.  I also had it happen a couple times where the speed indicated in QLog would jump up high like 53 wpm.  It wouldn't send that fast, but it would show that speed.  Also the when I am sending with my paddle the characters I send are shown twice (echoed) in QLog.  I'm not sure if the could be in settings in my keyer or what.  But I also don't have a lot of expierence with it.  I bought it to use with my HL2 but stopped using that radio so the keyer went in the drawer.

foldynl, 2025-03-21:
It's strange that QLog works correctly with firmware 2020.05.27.01. With the latest firmware 2024.03.20.2239, it behaves as you describe. However, it doesn't seem to be a firmware issue because, without QLog, K3NG Winkey works correctly and also sends only one character via the Serial Port.

foldynl, 2025-03-21:
@mi0hoz. Are you using the original Winkey2 keyer or the K3NG keyer with Winkey2 emulation? It seems that K3NG changed the behavior of the speed pot in version 2024.03.15.1354 k3ng/k3ng_cw_keyer@6b9decc
, as noted in the comment: 'Changed potentiometer behavior to work correctly with N1MM+ pot settings.' This change means that the pot is ignored when the keyer is connected to the host (PC).

>>> I also had it happen a couple times where the speed indicated in QLog would jump up high like 53 wpm.

@aa5sh I observe the same behavior, but it seems to be something in the firmware. I'll try going back version by version to see where it starts working.

foldynl, 2025-03-22:
@aa5sh : If `FEATURE_ROTARY_ENCODER` option is enabled together with `FEATURE_POTENTIOMETER` then K3NG Keyer increases the speed during sending. Disable `FEATURE_ROTARY_ENCODER` and everything will work. 

What I need to fix is ​​the speed pot setting. K3NG changed the speed setting behavior.

aa5sh, 2025-03-22:
Ok I’ll try changing that setting this evening when I get home.

aa5sh, 2025-03-22:
I am using a version of the K3NG keyed from Funtronics FK-11 that I bought off of eBay.  I went to their website www.elekitsorparts.com <http://www.eleitsorparts.com/> and grabbed the firmware from there to reload since it has the tweak version of the include file for this keyed.  It looks like it has just the ROTARY_ENCODER enabled and not the POTENTIOMETER.  Do you want me to try and flip that and rebuild and load it?

aa5sh, 2025-03-22:
I did just build you bug fix version for WK and it does work fine now without the echoing of characters and the speed did track both ways with the potentiometer.

foldynl, 2025-03-23:
@aa5sh  many thanks for testing. 

>>> Do you want me to try and flip that and rebuild and load it?

it depends on you. I think that if you don't have a rotary sensor connected (it is NOT a speed pot), then this should be disabled. Otherwise there is unexpected behavior. But I only looked at the K3NG code very quickly.

commit e66856c7 (2025-03-22): Fixed #618 - WinKey2: fixed speed pot malfunction

K3NG changed the behavior of the speed pot in version 2024.03.15.1354
This change means that the pot is ignored when the keyer is connected
to the host (PC). Therefore QLog sends "Set Speed" command when
the speed pot information is received.
closed 2025-04-04 by foldynl