foldynl/QLog #622: call e lotw

ik3ocd, 2025-03-21:
good morning, I think it's a little problem, I can't remove an old call that I had created with a special activity call, this only happens in the lotw part, I'll start by saying that I also deleted the previously created profile ii3icz, this means that I have to remember to select the main callsign every time during the lotw download phase, can you help me ladislav how to remove the call ii3icz which is no longer usable, thanks and 73 from ik3ocd flavio

[image]

[image]

only ik3ocd should remain

aa5sh, 2025-03-21:
That selection box is pulled directly from your log.  It looks for unique entries in this field:

[image]

So the only way to remove it would be to edit those entries to delete them or change the call sign.  You can create a user filter to find them list this:

[image]

ik3ocd, 2025-03-21:
Hi, thanks for replying, I did what you advised, but unfortunately the problem persists on the services as previously stated on lotw, the special call used previously under my callsign remains, on my side I have what was possible, maybe I'm still missing something. I await your advice, thanks.

ik3ocd, 2025-03-21:
thanks again for the help but this does not solve it for me., I wonder if it is possible to manually edit the specific configuration file inside the qlog dir and intervene manually. I looked for something that alluded to II3ICZ but without success.

foldynl, 2025-03-21:
@ik3ocd .... so again...the box contains values from the database `station_callsign`. The only thing I see here is that the state of this combo is only saved but not set. But that has nothing to do with what the content of this select box is. You should have the same content of the combo box for `Upload`->`My Callsign` (`Mio Indicativo`)

I recommend you add `station_callsign` to your Logbook and go through all QSOs.