foldynl/QLog #624: Retrieving information on the station/s worded/confirmed for a given DXCC

iu0dhv, 2025-03-22:
Hello, when calling a station and having his/her call shown in the specific box, QLog usefully shows the situation for that given DXCC and whether that specific station has been worked before. 
What is not possibile however is to see who you actually worked/confirmed for that given DXCC band/mode. Would it be possible to have this information by clicking on the box for the given band/mode for which L/P/W is shown? 
Otherwise is there another shortcut to show this info?
73
Peter IU0DHV

foldynl, 2025-03-22:
Please provide an example of why I would like to see this DXCC detail during the QSO. 

This feature is already implemented in Logbook -> Awards. If you double click on the work/confirmed then the main Logbook displayed QSO for particular DXCC.

iu0dhv, 2025-03-22:
One exemple that comes to my mind is for instance the following. Let's say that I have LZ worked on several bands but, albeit having uploaded all QSOs on LOTW, only on two bands I got LZ confirmed. Now, I see that LZ5GGG and LZ1HHH (fantasy calls) are on air on one of the bands where I am still missing confirmation and I would like to know with one click if any of them is the station that I have worked and that did not confirm on LOTW. If this were the case, I would not bohter to call them again. If not I might be interested in calling. 
Indeed I could type both calls in the callsing box to see it they show up below, but it would also be efficient to see the information with one click on the colored box, as is the case for the award menu that you mentioned. 
best 73

foldynl, 2025-03-23:
I'm sorry, but for this example of use there is a table under DXCC - Station Statistics. How correctly you wrote it is necessary to enter the Callsign, but that makes more sense to me than clicking on the DXCC entities. From this there is Awards, where you can click on different Entities and see.