foldynl/QLog #633: Incorrect Field Mapping for Received eQSL Messages

dimarmnes, 2025-03-26:
I am developing a mini application (almost a script) for my electronic QSL cards. I noticed that Qlog exports the message received in an eQSL in the QSLMSG field. The QSLMSG field should be for the messages that I want to include in my QSLs, and the received message along with the eQSL should be indicated in the QSLMSG_RCVD field.

73, LU1IDC Diego

aa5sh, 2025-03-27:
> dimarmnes
>  created an issue

I’m trying to make sure I understand.  It looks like currently when QLog processes EQSL confirmations it stores what EQSL sends in the QSLMSG field in QSLMSG in the log.  But you are saying it should be put in the QSLMSG_RCVD field?  I am not very familiar with the ADIF spec.  I wouldn’t think that should be a big change just trying to make sure I understand.  

You are just doing a normal ADI export and that is where you are seeing this data?

foldynl, 2025-03-27:
@dimarmnes : QSLMSG_RCVD was introduced in ADIF 3.1.5. ADIF 3.1.5 was approved at the turn of November and December 2024. Yes, you're right, QSLMSG_RCVD should be used now. I was giving it some time until it became a commonly used field, but unfortunately, I forgot about it. Thanks for the reminder.

foldynl, 2025-03-27:
As I look at this, it's quite a dilemma how to do it correctly. Originally, QSLMSG was a general QSL field where QSL messages were exchanged. Now, this field represents sending. But older entries may contain messages that were received. The problem is that I can't migrate the QSLMSG field to QSLMSG_RCVD because I don't know whether it contains sent or received messages.

What's even more interesting is that the QLog's QSO Detail should allow editing the QSLMSG field but shouldn't allow editing the QSLMSG_RCVD field. However, according to ADIF 3.1.4, the QSLMSG field can also contain QSLMSG_RCVD. I'm not sure if choosing to use QSLMSG in a different ADIF sense was a good move.

dimarmnes, 2025-03-27:
I hadn't noticed that QSLMSG_RCVD was recently added. And yes, it would be impossible to migrate fields from one to another for previous QSOs.

I've done some tests completing the QSLMSG field...
If I upload the QSO with QLog to eQSL, the message is NOT sent.
If I export the QSO and then upload it as an Adif in eQSL, the message is included.

What I have noticed is that when I download the Adif from eQSL, the messages sent by my contacts are in QSLMSG.

As you say, the QSLMSG field should be editable in the QSO Detail, but not the QSLMSG_RCVD field. And I understand that previous messages should stay where they are (impossible to reclassify). This is my interpretation if you want to adhere to ADIF rules.

aa5sh, 2025-03-27:
I would think it should be editable, what happens if I get a cool paper QSL card with a message or note I want to store in there?  This field isn’t exclusive to eQSL.  

I think it would be ok to just make the change so any new downloads the message would store or append to the field if not there?  Could also consider making a change similar to what we did to LoTW so if someone redownloads from eQSL it would place it there?

dimarmnes, 2025-03-27:
> I would think it should be editable, what happens if I get a cool paper QSL card with a message or note I want to store in there?  This field isn’t exclusive to eQSL.

It makes sense. Both fields should be editable.

foldynl, 2025-03-28:
@aa5sh : More changes related to this are included in testing_0.43.

- [NEW] - QSO Detail - QSLs/QSLr Msg is editable
- [CHANGED] - eQSL Upload - Added an option to choose a QSLMsg field
- [CHANGED] - eQSL Download - eQSL QSLMSG is stored to QLog QSLMSG_RCVD
- [CHANGED] - eQSL Download - Added QSLMSG_RCVD merging - eQSL message is appended

aa5sh, 2025-03-28:
Looks good to me.  I just downloaded some eQSLs and it stored the
messages in the appropriate fields now.   I also like the option for which
message to send.