foldynl/QLog #643: User Defined Rig Commands

n9gxa, 2025-04-08:
### Feature Description

I would like to request a way to add user-defined rig commands. I like to use them to adjust rig power to predefined levels with one quick click:

[image]

[image]

73 - Paul - N9GXA

foldynl, 2025-04-08:
Kyle summed up my opinion on this well in the [discussion](https://github.com/foldynl/QLog/discussions/640#discussioncomment-12744213).  

>>> In general the philosophy so far for QLog regarding cat integration is that the app is not a radio control app. It's a logging app and any radio integration should be directly tied to a logging feature requirement.

Rigs have more and more buttons, and users still want more and more buttons in the applications - why? The front panel should be the interface between operators and their rigs, not external application. The exception is remote operation but in this case QLog can never compete with specialized remote rig control applications

n9gxa, 2025-04-08:
Very well. It is a great program - Thank you.

As I become less flexible at times, I can find it difficult to turn a smaller, indented knob. While I am not operating remote and the rig is right in front of me, it can feel very remote at times. I try to operate at or under 50 watts until I come across that one operator that I think I just have to have logged and feel that I need to run 100 watts to get it. At times, it just isn't worth it to turn that knob to 100 w. Mouse clicking on a programmed button is so much easier. As I replied in the linked summation, after finding the band and mode drop-down on the rig widget, I was hoping this wasn't too far off the path.

It is not like this is going to stop me from using your program. Thank you for your reply.

aa5sh, 2025-04-08:
What Rig Control method do you use?  I use HamLib and run it against a
standalone running copy of RigCtlD so I can share rig control with QLog/
WSJTX/ and other apps.  I had thought about adding something like you
requested but do also understand about letting QLog do logging things.  But
I have written a program I run alongside QLog that handles talking to my
Steppir, Antenna Switch, rotor and SPE Amp.  I have a couple buttons that
I use to set the power like you to 50W or 100W.  I use a Yaesu FTDX-101 and
have started flipping between the SCU Lan-10 and a SCU 17 for data modes.
So I have buttons that change those menu items for me.  So I know it's
possible to send commands through hamlib/rigctld.  If you use something
like that I could try to make a simple app with 8 or so buttons you could
stuff cat commands in to send to RIgCtld.  Only question I would have is
which OS you are using.  I can test on Windows and MacOS (My default) but I
don't currently have any Linux.  I played around with something similar for
CW messages. I was having problems in QLog with messages looping and was
trying to see if the issue was QLog, Hamlib or what.  (Turned out to be me
causing it with my app polling rigctld to quickly and it was causing it to
loop)

Take out wanting to keep QLog focused, the hard thing to consider with
adding something like this is it would need to be integrated using HamLib,
OmniRig and TCI.  I know HamLib supports it but not sure how the other two
would support direct CAT commands.

n9gxa, 2025-04-08:
Thanks for the reply and motivation/direction. I did have QLog configured with Hamlib directly to the rig. After you mentioned rigctld, I remembered I had used Hamlib Net rigctl in the past. I reconfigured rigctld and changed QLog accordingly. I think I now have the basics setup and can change power via Linux terminal while QLog is running. I am looking for possible side-effects, but none found initially.

While not all in one program, I think this will help out nice. I just have to remember to switch back to QLog focus, but this way may add more flexibility.  Thanks again!!

n9gxa, 2025-04-09:
For now, I ended up using an xfce panel to hold the script launchers and also have keyboard shortcuts configured for each command. That leaves me more flexibility for any future additions. A nice feature about going this way does not take focus away from the logging window when changing power:

[image]

foldynl, 2025-04-10:
elegant solution.

closed 2025-04-10 by foldynl