foldynl/QLog #644: Rotator Widget QSO path

foldynl, 2025-04-11:
### Discussed in https://github.com/foldynl/QLog/discussions/642

<div type='discussions-op-text'>

<sup>Originally posted by **vk3ou** April  8, 2025</sup>
I wonder if, instead of the magenta desitnationAzimuthNeedle shown in the rotator widget, perhaps a qsoBearingNeedle would be better:
[image]
The bearing would have to be taken from NewContactWidget (I think - I am not sure). White for short path, grey for long path.
My thinking is that rather than show a destination (which is pretty much always = current rotator azimuth anyway), entering a new callsign in the NewContactWidget will automatically show the QSO paths in the rotator widget. 
Moving to short path or long path is then as easy as clicking the map where the needle points.

I am not sure if this idea has any merit?  What does everyone think?</div>

kyleboyle, 2025-04-13:
Something to think about. With this change there are now 2 static pointers on the map instead of one when the Callsign field is filled in. I was thinking that for our friends with less than perfect vision, colour blind, etc, it would be good to add an affordance to show which arrow represents the current rotator direction, and not rely on colour alone. example:

Before

[image]

After

[image]

It wasnt a big deal before because when there were 2 pointers on the map, one of them was moving so it was obvious which pointer represented to physical direction of the rotator.

```
    QPainterPath bigPath;
    bigPath.lineTo(-3, 0);
    bigPath.lineTo(0, -90);
    bigPath.lineTo(3, 0);
    bigPath.closeSubpath();
    antennaNeedle = compassScene->addPath(bigPath,
                                          QPen(QColor(0, 0, 0, 150)),
                                          QBrush(QColor(255, 191, 0), Qt::SolidPattern));

```

foldynl, 2025-04-15:
What do you think about reducing size of the QSO and Request Needles as wrote @vk3ou? I'm not sure if reducing the size will be OK with a small window.

[image]

or combination of both

[image]

and combination of both when QSO = Antenna

[image]

kyleboyle, 2025-04-15:
Different sizes works for me (the first one), it's still fairly obvious if I shrink those images down.

commit e0d3b8db (2025-05-09): Fixed #644 - Rotator Widget: Needle for QSO, requested azimuth and Antenna
closed 2025-05-09 by foldynl