foldynl/QLog #646: Runs in X11 session, core dump in Wayland session

n2nrv, 2025-04-12:
### What happened? How can it be reproduced? Provide as much information as possible.

When attempting to run QLog in a Wayland session, I get a core dump. I tried under an X11 session and QLog runs. I run EndeavourOS and installed via the AUR, which compiles the latest commit. The commit number is 8f9c758. It ran under Wayland at version 0.42.1 commit number 53d0d2b. I have an Nvidia card if that would make a difference. 

The errors:
Force XCB
13:18:15.617 [WARNING ] [0x751da4c84540] [default]                                          GBM is not supported with the current configuration. Fallback to Vulkan rendering in Chromium. [unknown:unknown:0]
13:18:15.672 [FATAL   ] [0x751da4c84540] [default]                                          GLX: Failed to find frame buffer configuration. [unknown:unknown:0]

### QLog Version

0.43.0

### Which OS are you using?

Linux

### QLog Installation Method

Self-Compiled

### Installation Details

uname -a: Linux xxxxxx 6.14.2-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 10 Apr 2025 18:43:59 +0000 x86_64 GNU/Linux
OS: EndevourOS

foldynl, 2025-04-12:
Could you please describe exactly under what circumstances the coredump occurs? Does it happen right after QLog starts up, or is there something specific you do to trigger it? 

Do I understand correctly that the issue did not occur in version 0.42.1?

n2nrv, 2025-04-12:
When using Wayland the coredump occurs immediately after the radio image comes up. The actual window does not appear at all. 
When using X11 no coredump occurs, QLog runs as expected.

> Do I understand correctly that the issue did not occur in version 0.42.1?

You are correct, the issue did not occur in version 0.42.1

foldynl, 2025-04-13:
what is your QT version? Did you change the QT version between 0.42.1 and 0.43?

n2nrv, 2025-04-13:
My current version of QT is 6.9.0. The version that was installed when compiling 0.42.1 was 6.8.2.

foldynl, 2025-04-14:
Qt 6.9 has been available since April 2, 2025. It's not wise to always follow the latest versions of this library. I've been burned several times doing that. I waited a long time for 6.8 to become usable, and even now it's not free of bugs – for example, the Windows version and OmniRig don't work properly under 6.8, and we still have to use auto-generated code from 6.3.

What I want to say is that QLog hasn't been tested with Qt 6.9, so I can't guarantee it will work correctly. However, I'd really appreciate it if you could send  your analysis of why it's crashing so we can adjust QLog accordingly.

clayton-workman, 2025-06-13:
For the record, I do not have this issue (coredump under Wayland). I am using QLog 0.44.1 self-compiled with Qt 6.9.1 on openSUSE tumbleweed.
 
[image]