foldynl/QLog #647: Temporarily Hold Comment Field Data

foldynl, 2025-04-12:
### Discussed in https://github.com/foldynl/QLog/discussions/645

<div type='discussions-op-text'>

<sup>Originally posted by **n9gxa** April 12, 2025</sup>
  Is there a way to keep the Comment field data temporarily? As an example, when I casually work other US State QSO Party events, I put "[State] QSO Party" in the comment field each time. Maybe a check box near the Comment field to keep the data for the next QSO and then uncheck when done working the event?

  So far, I like to have one logbook for all contacts, so I want to avoid creating a new logbook for different events.

Thanks,
73 - Paul - N9GXA</div>

kyleboyle, 2025-04-13:
Right click action menu for any field to toggle persistent value feature?
Would the user forget the feature is active?

foldynl, 2025-04-14:
These are exactly the questions I'm looking for answers to right now. I don't think right-clicking would be the solution, because the operator won't wait for it. In general, it seems to me that users forget about context menus.

The important question is whether this function is even necessary. I read your suggestion regarding ContestID. I had completely forgotten that it could be used for that purpose. For a long time, I used the Comment field as the QSL MSG for eQSL because WSJTX couldn’t send anything else. And WSJTX has a “retain” function for this field.

I’d leave this open for now — maybe we’ll come up with something later - but I think the main issue in the discussion has been resolved. Thanks.

closed 2025-07-18 by foldynl