foldynl/QLog #658: sending e-qsl lotw automatic or manual

ik3ocd, 2025-04-29:
### Feature Description

Hi, sometimes due to forgetfulness or laziness I forget to send the e-QSL, it would be nice to have the possibility to choose whether to send the e-QSL and the LOTW automatically at the end of the QSO or manually as in the clublog, I think it would help against forgetfulness. thanks and 73 from ik3ocd flavio

aa5sh, 2025-04-29:
I know @foldynl  is working on making some changes to the upload/download section for these items.  I'm not sure how feasible it would be to have it autoupload especially for LoTW without some significant changes.  I know for me for instance I have multiple locations setup in tQSL so each time I upload I am prompted to select which one to use.  Also it has a good bit more overhead with its signing process as compared to Clublog where it is a relatively simple webcall.  eQSL is fairly lightweight with simple webcall as well as QRZ.

foldynl, 2025-05-01:
Allow me to add my comment. As @aa5sh  wrote, I am currently reworking the entire upload/download functionality, but mentioned change is something different. I don’t think there’s something wrong to forget upload QSL to online services immediately. However, that’s just my opinion, and it’s not based on any concrete evidence.

What I can comment on is QLog’s behavior. If you upload QSL at the end, it can take a long time to close the program, and when an operator wants to exit the program, they usually don’t want to wait. Uploading at the beginning rises problems with the database, because QSLite DB doesn’t allow simultaneous writing of two things — this is a limitation on the SQLite side not QLog. 

Summary: I probably wouldn’t accept a change that performs QSL upload at exit. A change that uploads QSL at startup would require me to see how it’s implemented — then it might be possible to consider including something like that in the future.

But let me return to the beginning. I’m currently making a major change to the architecture of QSL/Upload/Download services. At this time, I’m not accepting any changes in this area, because it would create too many conflicts.

blhoward2, 2025-07-06:
I just switched from using MacLoggerDX and it uploads the contacts after each contact, in the background. It seems to work very well...