foldynl/QLog #660: data mode missing in rig control window

jmeloranta, 2025-04-30:
### What happened? How can it be reproduced? Provide as much information as possible.

The rig control window seems to report data mode on the radio as AM...

### QLog Version

0.0.0

### Which OS are you using?

Linux

### QLog Installation Method

Flatpak

### Installation Details

arch linux

aa5sh, 2025-04-30:
What radio are you using?

jmeloranta, 2025-04-30:
> [Tracking image][https://github.com/notifications/beacon/AJDFWHLPK22A4THPU2AJVS324AX43A5CNFSM6AAAAAB4EN5E6WWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTVJKDBRO.gif]

This is Elecraft K4.

foldynl, 2025-05-01:
to be honest, you didn't give much information. Please, at least describe what interface you are using? I assume Hamlib, what version? Are you  using Netrig/RigCtld? A screenshot of your rig settings in QLog would be useful. All this will speed up the solution to your problem

jmeloranta, 2025-05-01:
OK, sorry. This is hamlib with Elecraft k4 that has separate data mode (rather than USB). It shows as AM in the rig control window.

jmeloranta, 2025-05-03:
OK, I did some more testing (also screenshots attached). The AM mode seems to be there in the very beginning, so the menu setting is likely uninitialized. Changing bands will bring it to USB. The K4 separates between USB and DATA - not sure if other radios do this. But if they do, perhaps add data mode? Having it as USB rather than DATA will show wrong DXCC data (for phone rather than data) before the QSO. 

[image]
[image]

aa5sh, 2025-05-03:
If you notice right below the dropdown it is showing pktusb so it looks like it is getting the right mode from the radio.  Just for some reason it isn't matching to the dropdown.  Can you hit the dropdown for the mode and let me see what all options it shows.  I wonder if it has PKTUSB in there or not.  

It looks like QLog get the list of available modes from HamLib then tries to match the current mode to the list it gets in the combobox.  So I'm just curious what it shows for the K4 in there.

foldynl, 2025-05-03:
>>> Can you hit the dropdown for the mode and let me see what all options it shows. 

@jmeloranta  please, could you send us also the Setting dialog for your K4?  File->Settings->Equipment->Rigs and double click on k4 profile ? I have tested that the PKT Modes should be present for K4 in the QLog.

aa5sh, 2025-05-03:
I guess I never pay attention.  But I normally use a rigctld instance to connect to my radio so I can share QLog, WSJT and other software.  It seems to be doing the same thing if I connect through RigCtlD as reported above, where if I connect directly to the radio it shows fine.  With RigCtlD it only returns a few modes:

[image]

If I run 'rigctl -m 1040 --dump-caps' in a terminal window I get these modes:

AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB FM-D FMN AMN PKTFMN

aa5sh, 2025-05-03:
I wonder if it has something to do with this?

[image]

jmeloranta, 2025-05-03:
> I wonder if it has something to do with this?
>
> image.png (view on web)[https://github.com/user-attachments/assets/2b3ab97f-8fef-42d6-90cb-156ab1bb4dff]

And, yes, I am using rigctld as well.

jmeloranta, 2025-05-04:
Here is the rig config. Another peculiarity is that it won't automatically connect to the rig. I have to enable connection manually every time I start qlog.

[image]

kyleboyle, 2025-05-04:
Auto connect: use the activity editor to configure auto connection at startup:

[image]

@aa5sh looks like you've found an enforced limitation for net rig connection type... I wonder if it is a workaround for a bug in a previous hamlib version.

foldynl, 2025-05-06:
If the Rigctld interface is active, then everything makes sense. I don't remember exactly, but I believe the NetRig interface doesn't have the 'Mode' defined until it's connected. So I probably used this static mode definition as a workaround. It seems that QLog needs to refresh the Mode combo after connection, just like it's done in case of TCI.

closed 2025-05-09 by foldynl