foldynl/QLog #661: Arrows next to frequency in the main window could be bigger

jmeloranta, 2025-04-30:
### Feature Description

The arrows for stepping up/down the frequency in the main window should be a bit larger. They are quite tricky to use on small screens and especially with touch screen. Also I often use my station using remote desktop from my cell phone and then it is really difficult to click on those arrows.

dl2ki, 2025-05-03:
As a consequence, all other menu arrows would also have to be enlarged, which would have a major impact on the overall design of the GUI. I often wonder whether this is necessary for individual cases with various extension requirements.

QLog is not designed to be operated with a cell phone. There are special remote apps for this, with a suitable GUI design.

foldynl, 2025-05-03:
I agree with @dl2ki . QLog has never been optimized for touch screen control. Operating via a touch display would require a completely different approach, as this kind of interaction is fundamentally different from using a keyboard and mouse. For example, it would involve a different style of context menus, a different button layout, and so on. If you need control via a mobile phone, I don't think QLog is the software you're looking for. I would rather recommend WaveLog or Cloudlog.

jmeloranta, 2025-05-03:
> [Tracking image][https://github.com/notifications/beacon/AJDFWHLBCN2HXUZERUWX5TT24T4KPA5CNFSM6AAAAAB4EN7MOOWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTVJZQJIS.gif]

It works fine except that it just takes extra effort with those arrows. It is also with laptop using a trackpad. Easy to miss. Anyway, if it would affect the whole UI then there is no point...

kyleboyle, 2025-05-04:
@jemlorata, perhaps an alternative input approach you could try when you are using a small screen with a keyboard: use the keyboard up and down arrows when that field is focused.

closed 2025-05-24 by foldynl