foldynl/QLog #666: Seg Fault loading qlog_0.44 on 24.04

alphafox02, 2025-05-10:
### What happened? How can it be reproduced? Provide as much information as possible.

Sorry for being lite on details, quickly checked the latest qlog .44 release on 24.04 ( qlog_0.44.0-1~noble_amd64.deb). 

Upon loading Qlog and making it past the initial loading screens, it crashes when it shows the radio back to terminal.

If I change nothing else expect go grab the .43 release for Noble, it loads and runs without seg fault. I started looking at debugging. 

### QLog Version

0.0.0

### Which OS are you using?

Linux

### QLog Installation Method

DEB/RPM

foldynl, 2025-05-10:
I think that I will need more info. I have tested the 0.44 on Kubuntu 24.04 (PPA) without any issue. My configuration is shown below. It would be good to provide the coredump file if it is possible or send a debug log file - more info about the debug file on [Wiki](https://github.com/foldynl/QLog/wiki/Debug-Log-Level)

[image]

foldynl, 2025-05-10:
please, would you also provide information about your Qlog's rig setting. Do you use the auto-connect feature.?

alphafox02, 2025-05-10:
I may have missed a bit, I'll get it into a better capture. This is back to the .44 deb on 24.04.

`13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   do layout [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFrame (0--18), parent=0x0 [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFrame (0--18), parent=0x0 [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutStruct: x_left 0 x_right 0 fullLayout true [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFlow from= 0 to= 18 [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutBlock from= 0 to= 18 ; width 0 (maxWidth is btw 120.422 ) [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   do layout [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFrame (0--18), parent=0x0 [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFrame (0--18), parent=0x0 [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutStruct: x_left 0 x_right 100 fullLayout true [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFlow from= 0 to= 2147483647 [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutBlock from= 0 to= 2147483647 ; width 100 (maxWidth is btw 120.422 ) [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   do layout [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFrame (0--18), parent=0x0 [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFrame (0--18), parent=0x0 [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutStruct: x_left 0 x_right 0 fullLayout true [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFlow from= 0 to= 2147483647 [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutBlock from= 0 to= 2147483647 ; width 0 (maxWidth is btw 120.422 ) [unknown:unknown:0]
13:50:25.048 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   do layout [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFrame (0--18), parent=0x0 [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFrame (0--18), parent=0x0 [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutStruct: x_left 0 x_right 100 fullLayout true [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFlow from= 0 to= 2147483647 [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutBlock from= 0 to= 2147483647 ; width 100 (maxWidth is btw 120.422 ) [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   do layout [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFrame (0--18), parent=0x0 [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFrame (0--18), parent=0x0 [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutStruct: x_left 0 x_right 0 fullLayout true [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFlow from= 0 to= 2147483647 [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutBlock from= 0 to= 2147483647 ; width 0 (maxWidth is btw 120.422 ) [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   do layout [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFrame (0--18), parent=0x0 [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFrame (0--18), parent=0x0 [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutStruct: x_left 0 x_right 100 fullLayout true [unknown:unknown:0]
13:50:25.049 [DEBUG   ] [0x70ad1dc89a40] [qt.text.layout]                                   layoutFlow from= 0 to= 2147483647 [unknown:unknown:0]
[... cut ...]
13:50:25.171 [DEBUG   ] [0x70ad1dc89a40] [qlog.ui.rotatorwidget.function.entered]           *** [void RotatorWidget::setUserButtonDesc(QPushButton*, const QString&, double):ui/RotatorWidget.cpp:323]
13:50:25.171 [DEBUG   ] [0x70ad1dc89a40] [qlog.ui.rotatorwidget.function.entered]           *** [void RotatorWidget::setUserButtonDesc(QPushButton*, const QString&, double):ui/RotatorWidget.cpp:323]
13:50:25.171 [DEBUG   ] [0x70ad1dc89a40] [qlog.ui.rotatorwidget.function.entered]           *** [void RotatorWidget::rotDisconnected():ui/RotatorWidget.cpp:500]
Segmentation fault (core dumped)
`

alphafox02, 2025-05-10:
Sorry, that's a massive copy/paste. As for the second question you asked, I've only tried to load qlog from terminal. I'm unable to get to any of the settings as it doesn't fully load with the .44 build.

alphafox02, 2025-05-10:
Not sure if this bt from gdb helps any

`Thread 1 "qlog" received signal SIGSEGV, Segmentation fault.
0x00007ffff6ca633b in QGraphicsItem::setVisible(bool) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
(gdb) bt
#0  0x00007ffff6ca633b in QGraphicsItem::setVisible(bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#1  0x000055555596ab0a in ??? ()
#2  0x000055555596fb06 in ??? ()
#3  0x00005555558ccaa1 in ??? ()
#4  0x00005555558ba9f7 in ??? ()
#5  0x000055555561b465 in ??? ()
#6  0x00007ffff502a1ca in __libc_start_call_main (main=main@entry=0x555555619e90, argc=argc@entry=1, argv=argv@entry=0x7fffffffde88)
    at ../sysdeps/nptl/libc_start_call_main.h:58
#7  0x00007ffff502a28b in __libc_start_main_impl
    (main=0x555555619e90, argc=1, argv=0x7fffffffde88, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffde78)
    at ../csu/libc-start.c:360
#8  0x0000555555628b15 in ??? ()
(gdb) 
`

alphafox02, 2025-05-10:
[qlog_debug_20250510135814.log](https://github.com/user-attachments/files/20140834/qlog_debug_20250510135814.log)

mi0hoz, 2025-05-10:
QT_LOGGING_RULES=qlog.*=true qlog
19:05:04.037 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.function.entered]                  *** [void setupTranslator(QApplication*, const QString&, const QString&):core/main.cpp:49]
19:05:04.037 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.function.parameters]               "" "" [void setupTranslator(QApplication*, const QString&, const QString&):core/main.cpp:51]
19:05:04.037 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.runtime]                           Looking for a translation in "/usr/local/bin" "i18n/qlog_ga_IE" [void setupTranslator(QApplication*, const QString&, const QString&):core/main.cpp:110]
19:05:04.037 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.runtime]                           Looking for a translation in "/home/daddy/.local/share/hamradio/QLog" "i18n/qlog_ga_IE" [void setupTranslator(QApplication*, const QString&, const QString&):core/main.cpp:110]
19:05:04.038 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.runtime]                           Looking for a translation in "/usr/share/ubuntu-xorg/hamradio/QLog" "i18n/qlog_ga_IE" [void setupTranslator(QApplication*, const QString&, const QString&):core/main.cpp:110]
19:05:04.038 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.runtime]                           Looking for a translation in "/usr/share/gnome/hamradio/QLog" "i18n/qlog_ga_IE" [void setupTranslator(QApplication*, const QString&, const QString&):core/main.cpp:110]
19:05:04.038 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.runtime]                           Looking for a translation in "/usr/local/share/hamradio/QLog" "i18n/qlog_ga_IE" [void setupTranslator(QApplication*, const QString&, const QString&):core/main.cpp:110]
19:05:04.038 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.runtime]                           Looking for a translation in "/usr/share/hamradio/QLog" "i18n/qlog_ga_IE" [void setupTranslator(QApplication*, const QString&, const QString&):core/main.cpp:110]
19:05:04.038 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.runtime]                           Looking for a translation in "/var/lib/snapd/desktop/hamradio/QLog" "i18n/qlog_ga_IE" [void setupTranslator(QApplication*, const QString&, const QString&):core/main.cpp:110]
19:05:04.038 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.runtime]                           Looking for a translation in QLog's resources [void setupTranslator(QApplication*, const QString&, const QString&):core/main.cpp:126]
19:05:04.038 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.runtime]                           Cannot find any translation file [void setupTranslator(QApplication*, const QString&, const QString&):core/main.cpp:141]
19:05:04.038 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.appguard.function.entered]              *** [AppGuard::AppGuard(const QString&):core/AppGuard.cpp:45]
19:05:04.038 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.appguard.function.entered]              *** [bool AppGuard::tryToRun():core/AppGuard.cpp:94]
19:05:04.038 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.appguard.function.entered]              *** [bool AppGuard::isAnotherRunning():core/AppGuard.cpp:74]
19:05:04.051 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.function.entered]                  *** [void createDataDirectory():core/main.cpp:145]
19:05:04.051 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.runtime]                           "/home/daddy/.local/share/hamradio/QLog" [void createDataDirectory():core/main.cpp:148]
19:05:04.075 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.function.entered]                  *** [bool openDatabase():core/main.cpp:156]
19:05:04.076 [DEBUG   ] [0x7bc11fc6da40] [qlog.data.data.function.entered]                  *** [static QString Data::dbFilename():data/Data.cpp:631]
19:05:04.077 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.main.runtime]                           Pragma result: "wal" [bool openDatabase():core/main.cpp:183]
19:05:04.077 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.migration.function.entered]             *** [static bool Migration::backupDatabase(bool):core/Migration.cpp:85]
19:05:04.077 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.logparam.function.entered]              *** [static QVariant LogParam::getParam(const QString&, const QVariant&):core/LogParam.cpp:53]
19:05:04.077 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.logparam.function.parameters]           "last_backup" [static QVariant LogParam::getParam(const QString&, const QVariant&):core/LogParam.cpp:55]
19:05:04.077 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.logparam.runtime]                       Waiting for mutex [static QVariant LogParam::getParam(const QString&, const QVariant&):core/LogParam.cpp:57]
19:05:04.077 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.logparam.runtime]                       Using logparam [static QVariant LogParam::getParam(const QString&, const QVariant&):core/LogParam.cpp:57]
19:05:04.077 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.logparam.runtime]                       value:  QVariant(Invalid) [static QVariant LogParam::getParam(const QString&, const QVariant&):core/LogParam.cpp:92]
19:05:04.077 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.migration.runtime]                      The last backup date QDate(Invalid) Force false [static bool Migration::backupDatabase(bool):core/Migration.cpp:93]
19:05:04.077 [DEBUG   ] [0x7bc11fc6da40] [qlog.core.migration.runtime]                      (QFileInfo(/home/daddy/.local/share/hamradio/QLog/qlog_backup_20250510.adx)) [static bool Migration::backupDatabase(bool):core/Migration.cpp:124]
[... cut ...]
19:05:04.411 [DEBUG   ] [0x7bc11fc6da40] [qlog.ui.rotatorwidget.function.entered]           *** [void RotatorWidget::setUserButtonDesc(QPushButton*, const QString&, double):ui/RotatorWidget.cpp:323]
19:05:04.411 [DEBUG   ] [0x7bc11fc6da40] [qlog.ui.rotatorwidget.function.entered]           *** [void RotatorWidget::setUserButtonDesc(QPushButton*, const QString&, double):ui/RotatorWidget.cpp:323]
19:05:04.412 [DEBUG   ] [0x7bc11fc6da40] [qlog.ui.rotatorwidget.function.entered]           *** [void RotatorWidget::rotDisconnected():ui/RotatorWidget.cpp:500]
Segmentation fault (cuimhne dumpáilte)

alphafox02, 2025-05-10:
I did git checkout 20183e8 (prior to it talking about rotator changes)

Built from source, Qlog launches without a seg fault.

foldynl, 2025-05-10:
This commit is actually 0.43. If you are reporting the the 0.43 is running without issue then the commit is the same code.

foldynl, 2025-05-10:
Are you starting QLog for the first time?

alphafox02, 2025-05-10:
Yes, QLog was never ran on the computer before. I started with the .44 deb first or it may have been the latest source, can't recall.

mi0hoz, 2025-05-10:
Yes I'm starting for first time

alphafox02, 2025-05-10:
The very next commit, 9a06ac8, is when the segfault starts. Makes sense what you said about 2018e8. Maybe it's just something unique missing in Lubuntu?

foldynl, 2025-05-10:
This code is problematic - https://github.com/foldynl/QLog/blob/f4ca2be169ac150d023bc4fc73e3dffdbcdc96f8/ui/RotatorWidget.cpp#L511

If you are running QLog for the first time, there is the sequence, when antennaNeedle is null. I am going to prepare the fix.

alphafox02, 2025-05-10:
I think I figured out a workaround. I started .43, made a fake station profile as well as filled in a profile name on the rotator tab. I saved it. I then opened up .44 I built from source and it loaded.

alphafox02, 2025-05-10:
I see, so it's just because it's a first time run. I'm just getting around this by starting the .43 to load initial settings I suppose.

mi0hoz, 2025-05-10:
Fixed, thanks

alphafox02, 2025-05-10:
Patch works great, thank you!

commit 7325245c (2025-05-11): Fixed #666 - SegFault Rotator Widget - Quick and dirty fix - clean it up later
closed 2025-05-11 by foldynl