foldynl/QLog #667: QLog already running

mi0hoz, 2025-05-10:
### What happened? How can it be reproduced? Provide as much information as possible.

I'm running Ubuntu 24.04 and install QLog from the PPA. It wouldn't run. It would seg fault and core dump. I uninstalled it and downloaded the source and compiled it. Now it says QLog is already running and won't launch. I've looked in /var/lock but can't find the lock file.

### QLog Version

0.0.0

### Which OS are you using?

Linux

### QLog Installation Method

Self-Compiled

foldynl, 2025-05-10:
The lock is not present in /var/lock but in the shared memory. Please, try to reboot or I think logging out and back in sometimes helped.

mi0hoz, 2025-05-10:
~$ /usr/local/bin/qlog
18:49:00.040 [WARNING ] [0x73b73c24ea40] [default]                                          Cannot prepare select parameter statement for "last_backup" [static QVariant LogParam::getParam(const QString&, const QVariant&):core/LogParam.cpp:74]
18:49:00.061 [WARNING ] [0x73b73c24ea40] [default]                                          Cannot prepare select statement [long int LogFormat::runExport():logformat/LogFormat.cpp:1057]
18:49:00.061 [WARNING ] [0x73b73c24ea40] [default]                                          Cannot prepare insert parameter statement for parameter "last_backup" [static bool LogParam::setParam(const QString&, const QVariant&):core/LogParam.cpp:31]
18:49:00.061 [WARNING ] [0x73b73c24ea40] [default]                                          Cannot prepare select parameter statement for "last_backup" [static QVariant LogParam::getParam(const QString&, const QVariant&):core/LogParam.cpp:74]
18:49:00.062 [WARNING ] [0x73b73c24ea40] [default]                                          Cannot prepare select statement [long int LogFormat::runExport():logformat/LogFormat.cpp:1057]
18:49:00.062 [WARNING ] [0x73b73c24ea40] [default]                                          Cannot prepare insert parameter statement for parameter "last_backup" [static bool LogParam::setParam(const QString&, const QVariant&):core/LogParam.cpp:31]
Segmentation fault

foldynl, 2025-05-10:
You should run qlog in the debug mode:

```
QT_LOGGING_RULES=qlog.*=true ./qlog
```
Ideally redirect the output to a file. Then zip it up and send it. It will probably be large.

foldynl, 2025-05-10:
sorry. the command should be:
```
QT_LOGGING_RULES=qlog.*=true qlog
```

alphafox02, 2025-05-10:
I think we're both working on the same thing. 

https://github.com/foldynl/QLog/issues/666

foldynl, 2025-05-10:
Are you starting QLog for the first time?

foldynl, 2025-05-10:
the same issue as #666. I am closing it.

closed 2025-05-10 by foldynl