foldynl/QLog #671: QLog only accepts 126 contact entries

watchdoctor, 2025-05-13:
### What happened? How can it be reproduced? Provide as much information as possible.

I have QLog installed on ubuntu 25.04 and the logging program will only accept 126 contacts.  At 127, when I enter a contact and hit enter it does not show up as a saved contact.  I've tried manual entry and rebooting restarting.  Can someone help me understand and resolve why the limit is 126 contacts?

### QLog Version

0.0.0

### Which OS are you using?

Linux

### QLog Installation Method

DEB/RPM

aa5sh, 2025-05-13:
I'm not sure what could be going on.  I know very little about Linux.  But wanted to see if you could post a screenshot of the Main QLog window?

watchdoctor, 2025-05-13:
[image]

watchdoctor, 2025-05-13:
I don't see an easy way to post a screenshot here.

foldynl, 2025-05-13:
send me an email with the screenshot. I will post it here.

kyleboyle, 2025-05-16:
Also make sure that there are no filters active for the log.

[image]

PS you can upload a screenshot by simply dragging the image file into the message editing area

watchdoctor, 2025-05-17:
[image]

watchdoctor, 2025-05-17:
Nothing with the filters seemed to have any impact on entering QSO number 127.  127 will just fail and not enter into the log.  There is also no feedback to tell me any information.  When I hit save it just does nothing.

dimarmnes, 2025-05-17:
You have the band filter activated, are you sure you have tried changing it?

Diego Carratini

aa5sh, 2025-05-17:
Not sure what band you are working but this would only show stations for 20M.

[image]

watchdoctor, 2025-05-17:
well, you are all correct.  When I put it to "band" which I guess is any band I can see the contacts.  Look like no bug and pilot error on my part

watchdoctor, 2025-05-17:
Closing Issue.  Pilot Error

closed 2025-05-17 by watchdoctor