foldynl/QLog #673: Contacts that come in from WSJTX don't upload to LOTW

watchdoctor, 2025-05-19:
### What happened? How can it be reproduced? Provide as much information as possible.

When QLog receives contacts from WSJT-X they will get queued up for LOTW but when I use 'show selection' from the LOTW menu no contacts will show up.  If I manually enter a qso they will show up as ready to send to LOTW.  This only happens with QSO's that come in from WSJT-X.  I can't get them to ever be in a state that I can upload to LOTW.

[image]

### QLog Version

0.0.0

### Which OS are you using?

Linux

### QLog Installation Method

DEB/RPM

watchdoctor, 2025-05-19:
[image]

watchdoctor, 2025-05-19:
After some research I think I found out the issue.  QLog is set with my grid square as EM00, WSJT-X is set with my grid square EM00wf.  So this is maybe not a bug but a way that grid squares are handled with LOTW.  I see two selections in the drop down to send QSO's to LOWT.  One is EM00, one is EM00wf.  Where does QLog pull these options from?

aa5sh, 2025-05-19:
> watchdoctor

In QLog station profile you would enter your Gridsquare and that is used if you log directly in QLog.  For the WSJT it uses whatever you enter there.  For me since I mainly do HF in WSJT I just use my 4 digit grid to make it easier and not the 6 digit so I have to change it when uploading.

But you also need be mindful when uploading to the other services to check both grids in the dropdown.

Michael, AA5SH

watchdoctor, 2025-05-19:
Yes, just a sneaky little issue if the grid square from WSJT-X is a 7 character and your station profile is a 4 character.  This should be a documentation issue around wsjt-x / QLog interfacing

closed 2025-05-19 by watchdoctor