foldynl/QLog #677: Add JS8 to legacy_modes.json

macifell, 2025-06-02:
### Feature Description

I am trying to write an application for getting JS8Call logs imported automatically into QLog. Since QLog only handles WSJT-X and fldigi specifically, the best way seems to be to convert the output from JS8Call into the format that would be expected from WSJT-X. This is working great, but a problem is that JS8 is not listed in the legacy_modes.json file. Unless it is added there, I cannot use the same two-step process that WSJT-X uses to send over logs. This would be ok... except it seems that the values for qsl_sent, lotw_qsl_sent, and eqsl_qsl_sent do not get set correctly if only the adif log is sent over and not the initial "raw" log.

I am not asking for full JS8Call support, only for JS8 to be added to legacy_modes.json like so:
```
  "JS8": { "mode": "MFSK", "submode": "JS8" },
```

foldynl, 2025-06-03:
yes, the change makes sense. I will add it to the next release.

macifell, 2025-06-03:
@foldynl Thank you!

foldynl, 2025-07-11:
It is part of version 0.45.

closed 2025-07-11 by foldynl