foldynl/QLog #678: Setting to enter path to backup folder

w0rld, 2025-06-07:
### Feature Description

I want to be able to automatically backup to an external drive. This would allow backup to another drive rather than the machine drive.

w0rld, 2025-06-09:
> At first glance, this solution seems relevant, but there are several limitations, why I haven't implemented it yet.
>
>   1.  The use case for backup is that some backup software, not QLog, takes care of the backup. For everything to work, QLog must produce backup data to the path that is usually backed up - the user's home.
>
>   2.  QLog does not actually perform backups. It only performs exports. It depends on the user how use these files.
>
>   3.  QLog performs backups regularly, but also irregularly. However, backups must always be performed in cases of "irregular" backups. What should QLog do if the backup drive is not present - not start?
>
> Points 1 and 2 are more about mindset, but how do you think QLog should behave in point 3?

In #1, the user could provide a backup path in QLog into their home directory and then their backup process would take care of backing it up.

In #3, if the backup drive is not available, the program should just throw up an informative dialog letting the user know "No backup has been created because the backup path is not available. Please fix the backup path in the settings." and then not do anything. This is a typical process for a lot of other programs.

Several other logging programs do not automatically backup during usage, and only offer to backup when the program is closed. They also offer a manually triggered backup at any time.

Jim, W0RLD

aa5sh, 2025-06-09:
For #3 it could export to the current location and prompt a message saying
it couldn't go to the user specified path and defaulted.

If a change is considered in these exports I would like it to go back to
when closing and potentially let the user define a frequency - even if at a
high level - (everytime, every 7 days etc.)

I know I don't have a huge log about 26,000 records but the exports are
pretty quick so I would prefer it do everytime I exit.

foldynl, 2025-06-10:
>>> the user could provide a backup path in QLog into their home directory and then their backup process would take care of backing it up.

QLog already saves all backup files to home. It does not require user settings to do so.

> if the backup drive is not available, the program should just throw up an informative dialog letting the user know "No backup has been created because the backup path is not available. Please fix [...]

To understand why this is happening, you need to look into the code. Put simply, I need to be 100% sure that the backup completes successfully at startup, because it's done before the database migrations. If a migration fails for any reason, at least the user will still have a complete backup of their log.

If we introduce any additional logic at this point, it only increases the chances of something going wrong—user errors, and so on. And that's exactly what I want to avoid. At this step—backup before migration—I want to be absolutely sure that the user cannot lose any data, under any circumstances.

closed 2026-10-20 by foldynl