foldynl/QLog #679: Severe lag using hamlib/flrig

csibbitt, 2025-06-07:
### What happened? How can it be reproduced? Provide as much information as possible.

First of all thanks for this excellent software; it is a joy to use. 

I recently migrated from 0.39 on windows to 0.43/44 on linux and ran into this severe lag problem controlling my ft991a via flrig (same setup works fine in windows).

Another user has also reported this bug here: https://github.com/foldynl/QLog/discussions/625

My setup:
* Ubuntu 24.04
* Qlog 0.44.1-1~noble from your ppa
	* Also tested 0.39.0 since it works in windows, but it's also laggy
* libhamlib4t64 4.5.5-3.2build2 (ubuntu repo)
* flrig 2.0.04-1build2 (ubuntu repo)

Here's how I can reproduce the problem:

* Confirm Flrig can PTT reliably on and off many times with no perceptible lag
* Connect Qlog to Flrig via Hamlib, FLRig (20221109.0), Network (localhost). Enable the top 6 Rig Features checkboxes up to "PTT State". Poll interval 500ms.
* VFO and mode changes on the rig show up quickly and reliably in Qlog at first
* I set up Wireshark to watch the connection
* See Qlog sends the get_vfoA command ~once/1s
* Use Qlog PTT/band change commands several times
	* At first they respond quickly 
	* Soon degrade to multiple seconds of lag
	* Sometimes commands are just lost
* See Qlog sends the get_vfoA command ~once/7s
* VFO and mode changes show up with considerable lag in Qlog
* Using PTT is a dangerous gamble - the timing of the key-down is very late and unpredictable, sometimes the PTT-off is missed entirely
* Controls still work perfectly via flrig, hamlib's rigctl

Adjusting the poll interval seemed to help (for awhile?). The default is 500ms, but going to 1000ms made the lag basically go away. If I hammered on the commands really fast it would still screw it up, though. This workaround hasn't proved reliable over time, but does occasionally seem to help.

When using hamlib/serial config, the problem does not occur, so I've only reproduced it in hamlib/flrig so far.

I tried taking QLog out of the equation and testing just hamlib->flrig using the CLI tool `rigctl`, like this ('-m 4' means "flrig", 'T' is PTT): `for i in {1..25}; do rigctl -m 4 -r 127.0.0.1:12345 T 1; sleep 0.25; rigctl -m 4 -r 127.0.0.1:12345 T 0; sleep 0.25; done` 

Running this rigctl loop several times in a row didn't produce the lag problem, so this suggests it might be a qlog bug.

### QLog Version

0.44.1

### Which OS are you using?

Linux

### QLog Installation Method

DEB/RPM

### Installation Details

Can also reproduce on my own builds or older versions

aa5sh, 2025-06-08:
Ok weird note for this one.  I typically use QLog on MacOS with SmartSDR for MacOS.  For some reason was having problems getting FLRig to talk with it. I did once and I noticed right off what you are saying with FLRig alone the PTT was as expected and we I point QLog to it there was inconsistent and slow.  

I was going to try and grab a Debug to look at the messages (https://github.com/foldynl/QLog/wiki/Debug-Log-Level) and I couldn't get FLRig to connect.  So I launched FLRig on a windows PC to a Windows SmartSDR instance and I didn't notice the PTT issues in QLog.

You may want to get a debug log.  You can filter to just the rig messages like this:
QT_LOGGING_RULES=qlog.rig.*.debug=true ./qlog

Michael, AA5SH

foldynl, 2025-06-08:
Hi, the debug output is not necessary in this case. I can replicate it.

closed 2025-07-11 by foldynl