foldynl/QLog #685: Problems entering my country to create my profile

JulioBerse, 2025-06-16:
### What happened? How can it be reproduced? Provide as much information as possible.

When I try to enter my country to create my profile, there is only a blue stripe and clicking to expand the box does nothing.
I am using a Brazilian distribution based on Manjaro. 

[image]

### QLog Version

0.44.1

### Which OS are you using?

Linux

### QLog Installation Method

Flatpak

### Installation Details

I use the Big Linux distribution, with DE Gnome 48. Installation via Flatpak in the distro's own program installer.

[image]

aa5sh, 2025-06-16:
I experienced this the other day when I was testing with running in a new namespace.  I played with it some the only way I could get the combobox to populate properly was by modifying this SQL statement in SettingsDialog.cpp taking out the locale parts.

[image]

Like this:

[image]

Not sure why it was messing up for me.  But I just tried it again and it did the same thing still.  The countries load fine in the NewContact widget and contact editor.  I thought it may have been something with me running it from qt creator.

Michael, AA5SH

aa5sh, 2025-06-17:
It is regular.  I even tried copying my normal database into the thew namespace directory and it will not load QSOs or the country doesn't show up in the settings dialog.

I tried to remove the different parts (translate_to_locale and LOCALEAWARE) and the only way it worked was to make it like above.  Doesn't really make sense to me what's going on.  Those functions work fine other places.

foldynl, 2025-06-17:
Can you test the patch below, if it changes the situation?

```
diff --git a/core/main.cpp b/core/main.cpp
index 5ae7ac6..0b3b1c3 100644
--- a/core/main.cpp
+++ b/core/main.cpp
@@ -219,8 +219,9 @@ static bool createSQLFunctions()
                                         case SQLITE_TEXT:
                                         {
                                             const char *text = reinterpret_cast<const char*>(sqlite3_value_text(argv[0]));
-                                            const QString &translatedText = QCoreApplication::translate("DBStrings", text);
-                                            sqlite3_result_text(ctx, translatedText.toUtf8().constData(), -1, SQLITE_TRANSIENT);
+                                            const QString translatedText = QCoreApplication::translate("DBStrings", text);
+                                            QByteArray utf8 = translatedText.toUtf8();
+                                            sqlite3_result_text(ctx, utf8.constData(), -1, SQLITE_TRANSIENT);
                                         }
                                             break;
                                         case SQLITE_NULL:
@@ -241,8 +242,8 @@ static bool createSQLFunctions()
                                      SQLITE_UTF16,
                                      nullptr,
                                      [](void *, int ll, const void * l, int rl, const void * r) {
-                                        const QString &left = QString::fromUtf16(reinterpret_cast<const char16_t *>(l), ll/2);
-                                        const QString &right = QString::fromUtf16(reinterpret_cast<const char16_t *>(r), rl/2);
+                                        const QString left = QString::fromUtf16(reinterpret_cast<const char16_t *>(l), ll/2);
+                                        const QString right = QString::fromUtf16(reinterpret_cast<const char16_t *>(r), rl/2);
                                         return QString::localeAwareCompare(left, right); // controlled by LC_COLLATE
                                      });
         }
```

aa5sh, 2025-06-17:
I'm not sure how to help get it replicated.  But yeah it happens when I create a new namespace.  I tried the patch and it didn't help.  I'm at a work school so can spend a little more time digging in this evening after class.  

The odd thing is it happens on this my work laptop but not on mine at home.  So not sure if it is some environmental variable not set or something.  Or could compare QT Versions or SQLite.

aa5sh, 2025-06-17:
Not sure if this is the issue but I see qt 6.9 has something listed with localization here:  https://doc.qt.io/qt-6/macos-issues.html but not sure if this also rolls over to Linux as well.

aa5sh, 2025-06-18:
Ok I did a little dirty debugging.  But it looks like for some reason the function isn't being created.  If I add the following code to the createSQLFunctions.  I get an error saying the function doesn't exist.  I do a quick test also to make sure the db connection is fine:

[image]

aa5sh, 2025-06-18:
One more thing I tried.  I ran 'SELECT * FROM pragma_function_list WHERE builtin = 0;' and it showed the 'translate_to_locale' function was in the db.  But if I turned around and tried to do a simple 'Select translate_to_locale(name) from dxcc_entities;' it says the function doesn't exist.

foldynl, 2025-06-20:
@JulioBerse . I may have found the cause of your issue. Based on the attached screenshot, it seems that you didn't have an internet connection. When QLog is launched for the first time, it requires an internet connection because it downloads various value lists from the internet — including the list of countries. If these lists aren't downloaded at the start, QLog will not function properly.

@aa5sh  - Michael, I don't know exactly where the issue is on your end. I’ve done several tests, and I must say that Qt 6.9, which you might be using on macOS, is a version that doesn’t work very well. However, that might not be your case. I also tried a few configuration options, but I always managed to retrieve the country list. 
Please check whether you might be using two different versions of libsql. You need to use the one from the Qt sources, not the default libsql from macOS. Otherwise, I really don't know what else it could be.

JulioBerse, 2025-06-23:
Atualizando ..... 73 fraternais a todos .

O Qlog está funcionando super bem , extremamente satisfatório o meu uso . Tudo funcionando inclusive o envio de QSOs para os programas externos . 
Usando em uma distro brasileira baseada no Manjaro , o Big Linux com Gnome .

foldynl, 2025-07-05:
@aa5sh : Lately, it seems to me that there are more and more bugs in Qt. I tested Qt 6.9 on Windows and QLog did not work correctly. There are many unexpected behaviors for me. I don't know if it's possible to stay on a specific Qt version on macOS, but I highly recommend  use Qt 6.8.

Regarding version 0.45: I’d appreciate it if you could test it. There have been many, many changes. I’ve been testing it myself for a few weeks now. I’d like to release it next week if everything goes well and I’m not tied up with more pressing matters, as has been the case for the past month and a half.

foldynl, 2025-07-14:
The main issue has been resolved. Closing.

closed 2025-07-14 by foldynl