foldynl/QLog #688: Activities "Save Arrangement" doesn't save the window width

FastThenLeft, 2025-06-20:
### What happened? How can it be reproduced? Provide as much information as possible.

I have two activities, "Home" and "Satellite".

Home looks like this:

[image]

Satellite looks like this:

[image]

When I switch from Home to Satellite, QLog doesn't resize the width of the Satellite window to what I saved previously:

[image]

Thank you again for such a great program!
-David, N9KT

### QLog Version

0.44.1

### Which OS are you using?

Linux

### QLog Installation Method

Flatpak

### Installation Details

Ubuntu 24.04 running X-Windows.

$ uname -a
Linux hamshack 6.11.0-26-generic #26~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Apr 17 19:20:47 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

aa5sh, 2025-06-21:
Just make sure after you position it the way you want it to hit Save Arrangement for each of your activities. It doesn't autosave.

FastThenLeft, 2025-06-21:
Yes, I did "Save Arrangement" after each change on each of the Activities. It still doesn't return the Satellite Activity to the same width as I set it to.