foldynl/QLog #695: New QSO, move between QSO Fields

dl2ki, 2025-07-06:
### What happened? How can it be reproduced? Provide as much information as possible.

Once a callsign has been entered in the 'New QSO' area, the cursor jumps to the centre digit of the 'RSTs' field after pressing 'TAB'. The same applies to the 'RSTr' field. This makes it necessary to adjust the cursor position if 'R < 5'. The same applies to the 'RSTr' field.

[image]

The behaviour only affects OMs who provide real RST reports. Whether to change the behaviour of the TAB key is up to the author. This may be common.  Personally I can also accept the current behaviour of the TAB key.

### QLog Version

0.44.1

### Which OS are you using?

Linux

### QLog Installation Method

DEB/RPM

foldynl, 2025-07-07:
How should this work effectively?
I know there have already been several discussions about this.
Should QLog jump to the first digi, and should the entire field be editable in insert mode (i.e., where digits are overwritten as you type)?

dl2ki, 2025-07-07:
A CW report consists of three digits: R, S and T. Consequently, you should also be able to enter three digits.

Assuming that R always equals 5, the current variant would be OK. Otherwise, the input cursor should jump to the first digit so that you can enter RST directly. This then requires the entry of three digits, as the RST system requires. Those who want to use the "599" default can simply jump to the next field with the TAB key.

In practice, '599' has apparently become the standard. It is often generated automatically and at high speed. This has nothing to do with the original purpose of the RST system. A report of R = 5 and S = 9 is especially unlikely with QRP portable stations (SOTA, POTA, WWF, etc.). Anyone can check this for themselves by listening in the CW band (or SSB band). However, in terms of sound quality (T), modern radio equipment hardly ever produces sound quality below 9.

Personally, I have got into the habit of deleting the RSTr and RSTs fields after pressing the tab key, so that I can then enter the three digits of a real report. So I have no problem leaving everything as it is.

dl2ki, 2025-07-09:
After some thought, I realised that deleting the RST defaults is actually the ideal solution for me.

Without the default values in the settings, you can enter the three digits as specified by the RST system. Alternatively, if you would prefer not to do this, you can use your own default values. In that case, it doesn't matter if the cursor jumps to the second digit.

So there is no need for me to make any changes.
Thanks again for the tip!

foldynl, 2025-07-10:
Thanks for your comment. I'll test some improvements, but it's not yet ready to be part of the next release.

dl2ki, 2025-07-30:
I just read your comment on commit "fe983df."

I'm always impressed by the simple, universal solutions you find for such minor issues. As always, great work, and an excellent idea!

foldynl, 2025-07-30:
I'm still not sure whether the overwrite mode is the right solution. Personally, I like it more than insert mode, but in some situations it feels a bit clumsy — for example, when you delete one or more characters, it leaves empty spaces behind. Have you tested it? How do you find working with the new overwrite mode?

I'm on vacation right now, so I can't fully focus on it, but I'd still be interested in hearing your opinion.

dl2ki, 2025-09-14:
Thank you for the new version. In my opinion, the use of the RST and RSTr fields now corresponds to the original concept of number allocation. Those who prefer to use standard settings will find that nothing has changed.

I happened to place the cursor on one of the fields and pressed the mouse button. A vertical black bar appeared, which can be moved 6 times in the field using the arrow keys on the keyboard, i.e. a total of 7 bars.

https://github.com/user-attachments/assets/c0d019ce-0785-4a08-b7be-e3d43d0332e3

foldynl, 2025-09-14:
Yes, that's right. Unfortunately, it's more complicated to explain, but this is a secondary effect of the Overwrite mode. There are pre-reserved places for RST, the behavior is different than for Insert Mode.

dl2ki, 2025-09-14:
Interesting.  

The black bar also appears when you jump into the field with the TAB key when the report default is 3 spaces.  

With a default of 3 digits, the cursor jumps correctly to the first digit.  

However, you can enter up to 5 digits, including ‘0’, in the field. Letters and special characters are also possible.

The basic function of the input fields is not significantly disrupted if you adjust to this. However, transferring the characters to the QSO list could be problematic. 

[image]

[image]

[image]

[image]

foldynl, 2025-09-14:
In general, `RST` fiels has format `xxxxxxx;`. It means that QT reserves 7 places for any character. The fact that any character can be entered is nothing new, it was already possible in previous versions. This version only introduced the limit that a maximum of 7 characters can be entered in overwrite mode. Why 7? Because `59+xxx` + 1 char reserve seemed to me to be the longest RST notation. It also has to be prepared for `-12` and other notations used in different modes. The fact that sometimes you may write `599` starting from position 2 doesn’t matter. The first position will be ignored, and the entry will start from the next position. Why not only digits? Because `+` and `-` can appear there. Why various characters and not only digits? Because in ADI, RST is defined as a [STRING](https://www.adif.org/315/ADIF_315.htm#QSO_Field_RST_RCVD). It’s not a number. Why 7 characters for all modes? Because I’m not able to define all operator requirements for `RST` across different modes. 

That’s, in short, the history of this field. So your examples are valid and expected. They were valid and expected even in previous versions. Is it good? I don’t know, I tried to make the input field as universal as possible. Introducing more restrictions seemed counterproductive to me.

dl2ki, 2025-09-14:
Thank you very much for the detailed explanation. This makes a lot of things clearer, so that we can adjust accordingly.

In principle, it's all understandable, but it certainly feels different for different users. Let's wait and see if there are any further inquiries about this.

We can leave it as it is for now.