foldynl/QLog #710: Non-standard characters are not show properly

oliverbross, 2025-07-24:
### What happened? How can it be reproduced? Provide as much information as possible.

I have found out if the operator uses any special characters, be it german, russion or others in the name, address, etc, they are not being shown correctly in the Qlog. 

Suggestion improvement - allow to show all characters properly. 

### QLog Version

0.45.0

### Which OS are you using?

Windows

### QLog Installation Method

EXE

foldynl, 2025-07-24:
Unfortunately, it's not clear from your description what you're reporting. A screenshot or more detailed description would be helpful.

oliverbross, 2025-07-24:
example of russian station:

[image]

qrz.com shows:

[image]

I have also seen the same problem with some German characters.

Hope it helps.

Interestingly, some Japanese Kanji are ok :-)

oliverbross, 2025-07-24:
another example from the cluster window:

foldynl, 2025-07-24:
Related to the Name field, please check whether you system supports (is correctly set) the UTF8 character set. QLog has no issue to show and store UTF-8 characters from QRZ Callbook as show screenshot below - tested under Windows 11 (QLog v0.45)

[image]

Related to the DXC: This is a known issue with the DX Cluster, where users can enter any character set and there is no character set information in the spots. Contributions and related pull requests are welcome if you have any idea how to fix it