foldynl/QLog #715: DX Cluster shows stations worked even though setting is not enabled

oliverbross, 2025-07-24:
### What happened? How can it be reproduced? Provide as much information as possible.

example:

[image]

Maybe the problem is on the operators side, however, one would expect to not to see station already worked appearing in the cluster ... it still happens. 

### QLog Version

0.45.0

### Which OS are you using?

Windows

### QLog Installation Method

EXE

foldynl, 2025-07-24:
discussed in #704 and via private email. This is not a QLog issue. We could consider Nice-to-Have feature request. 

Don’t worry — if I think something needs changing during a discussion, I’ll definitely open a feature request. I often do that right from the discussion/mailing list.

foldynl, 2025-07-24:
You will be warmly welcomed in the delevoper team ;-)

oliverbross, 2025-07-24:
I want to acknowledge the impressive work you’ve done with QLog—it’s clear you’re a highly skilled developer. While I’m not a programmer myself, I bring 30 years of IT experience to the table, including 20+ years in IT management, leading development teams, and overseeing large-scale projects—with a particular focus on UI/UX optimization.

My goal is simply to support QLog’s growth by offering well-considered suggestions from both a technical and user perspective. As an active ham operator and heavy logging software user, I regularly compare implementations across platforms and identify areas for improvement. I believe QLog has tremendous potential, and I’m happy to contribute ideas that align with its vision.

Looking forward to seeing where you take this project—and to continuing our productive discussions!

foldynl, 2026-11-10:
Statuses are redesigned. I close the issue.

closed 2026-11-10 by foldynl