foldynl/QLog #716: Deduplicate DX cluster spots to reduce clutter

oliverbross, 2025-07-24:
### Feature Description

Currently, when multiple operators spot the same station on the DX cluster, QLog displays each spot as a separate entry. This causes:

- Duplicate entries for the same station
- Cluttered DX cluster window
- Reduced visibility of unique spots
- Operator confusion about spot recency

Implement automatic deduplication with these behaviors:
1. **Replace existing spots** when a new spot arrives for the same callsign+frequency (±1kHz)
2. **Visual indicators**:
   - Brief highlight when a spot updates
   - Optional spot count badge (e.g., "KH1/K6ABC (3)")
3. **Configuration options** (in Settings > DX Cluster):
   - ☑ "Show only most recent spot per station" (default: on)
   - ☑ "Highlight updated spots" (default: on)
   - ☑ "Show spotter count" (default: off)

**Current Behavior**:
```
2023-12-20 14:05 ZL1ABC 14023.5 CW DXpedition (by N1XYZ)
2023-12-20 14:04 ZL1ABC 14023.5 CW DXpedition (by K2ABC)
2023-12-20 14:03 ZL1ABC 14023.4 CW DXpedition (by W3DEF)
```

**Proposed Behavior**:
```
2023-12-20 14:05 ZL1ABC 14023.5 CW DXpedition (by N1XYZ) [updated]
```

### Technical Considerations
- Dedupe key: `callsign + rounded frequency` (nearest 1kHz)
- Memory management: Remove oldest spots first when at capacity
- Performance: Hash table lookup for O(1) deduplication checks

### Additional Context
This matches behavior in:
- DXLab Suite Commander
- TrustedQSL (tqsl)
- Most modern contest logging software

Would greatly improve usability during:
- DXpeditions
- Busy contests
- Multi-op monitoring scenarios
```

Current real example from the DX Window:

<img width="713" height="62" alt="Image" src="https://github.com/user-attachments/assets/5fc4d0a0-1e61-41dc-9c7b-1a076e84f2c8" />

foldynl, 2025-07-24:
Oliver, please don't flood the forum with requests that are already implemented - this already exists in QLog - DXC Filter - Deduplication.

Please refer to the Wiki first, consider whether the request is beneficial to a wide range of people and then open an issue.

oliverbross, 2025-07-24:
Please, accept my apology, I forgot I had the d-dupe enabled:

[image]

I need to set ti up for more than 60 seconds.

Will be more careful next time. I don't want to waste your time and it even takes me time to write things up, so benefit is on both sides.

closed 2025-07-24 by oliverbross