foldynl/QLog #719: Fuzzy search for countries and regions

BI6NSL, 2025-07-31:
### Feature Description

Each time you need to find a country or region to view qso data, you have to search through the country and region list for a long time, and sometimes even need to search repeatedly. It is suggested to add an inputable item in the drop-down area to facilitate fuzzy search and make it easier for users to find

BI6NSL, 2025-07-31:
[image]

BI6NSL, 2025-07-31:
It is very inconvenient to find just one dxcc among hundreds of DXCCS or regions

aa5sh, 2025-07-31:
Are you talking about something like this:  

[image]

Where when I started typing in it filtered 'sta' to all countries with that in the name?

aa5sh, 2025-07-31:
I have a draft of  a PR to make it filter like that if @foldynl thinks it's a good idea to add it.

BI6NSL, 2025-07-31:
> 你在说这样的事情吗：
> 
> <img alt="图像" width="582" height="205" src="https://private-user-images.githubusercontent.com/84428382/472810708-074a9db4-b917-4008-b569-ef0ed7b02398.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTM5MzU4MzIsIm5iZiI6MTc1MzkzNTUzMiwicGF0aCI6Ii84NDQyODM4Mi80NzI4MTA3MDgtMDc0YTlkYjQtYjkxNy00MDA4LWI1NjktZWYwZWQ3YjAyMzk4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzMxVDA0MTg1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ5ZDlkMTQwMmNiNmE1NGFmMjQ0Zjg2Zjg2YWY5ZDFlN2Q3ZDFjMTg3ZDFjMDI2N2UxODI1MWJiN2E3NjI2MGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.3iUE5_jghbjsGyhqwkyFn6p7aqeaowecN_eaRNvPag4">
> 当我开始输入时，它在哪里过滤了“sta”到名称中带有该名称的所有国家/地区？

Yes, I think this will be very convenient

BI6NSL, 2025-07-31:
Thank you for your hard work. 73

closed 2025-07-31 by BI6NSL