foldynl/QLog #722: Repeated log recording

BI6NSL, 2025-08-04:
### What happened? How can it be reproduced? Provide as much information as possible.

For unknown reasons, it often happens that two identical logs are recorded. Please refer to the picture

[image]

[image]

[image]

### QLog Version

0.4.5

### Which OS are you using?

Windows

### QLog Installation Method

DEB/RPM

BI6NSL, 2025-08-05:
I use JTDX/WSJTZ to handle the FT8 work. After completing the QSO, I transfer the data to the GridTracker to achieve automatic QSO upload. Then, the GridTracker forwards the UDP data to the Qlog to achieve the final record.

I'm not sure if there is a problem with the GridTracker transmission.

Because in the past, after completing the QSO, JTDX/WSJTZ would transfer the data to Qlog via UDP to achieve local recording, and then Qlog would forward the UDP port data to the GridTracker to realize automatic QSO upload.
Everything is normal.

foldynl, 2025-08-07:
>>> Because in the past, after completing the QSO, JTDX/WSJTZ would transfer the data to Qlog via UDP to achieve local recording, and then Qlog would forward the UDP port data to the GridTracker to realize automatic QSO upload.
Everything is normal.

Does it mean, that if you use this setting eveythink is working as expected ?  Then I would recommend leaving it like this.

BI6NSL, 2025-08-07:
> > > > 因为过去，JTDX/WSJTZ在完成QSO后，会通过UDP将数据传输到Qlog，实现本地记录，然后Qlog会将UDP端口数据转发到GridTracker，实现QSO自动上传。一切都很正常。
> 
@> 这是否意味着，如果您使用此设置，则 everythink   按预期工作？那么我建议保持这样。

yes

foldynl, 2025-08-14:
Issue confirmed. C21TS provided more information. Details below from JTDX:

[image]

```
20250813_044145  -4  0.1 1404 ~ C21TS AC6SC DM14
20250813_044200.068(0)  Transmitting 18.1 MHz + 1855Hz  FT8:  AC6SC C21TS -04
20250813_044215  -4  0.1 1404 ~ C21TS AC6SC DM14
20250813_044230.066(0)  Transmitting 18.1 MHz + 1855Hz  FT8:  AC6SC C21TS -04
20250813_044245  -5  0.1 1404 ~ C21TS AC6SC DM14
20250813_044300.070(0)  Transmitting 18.1 MHz + 1855Hz  FT8:  AC6SC C21TS -05
20250813_044315  -5  0.1 1404 ~ W0DCO AC6SC DM14
20250813_044345  -4  0.1 1405 ~ W0DCO AC6SC R-07
20250813_044415  -5  0.1 1405 ~ W0DCO AC6SC 73            ^
20250813_044445  -5  0.1 1405 ~ C21TS AC6SC DM14
20250813_044500.066(0)  Transmitting 18.1 MHz + 1855Hz  FT8:  AC6SC C21TS -05
20250813_044515  -6  0.1 1405 ~ C21TS AC6SC DM14
20250813_044530.067(0)  Transmitting 18.1 MHz + 1855Hz  FT8:  AC6SC C21TS -06
20250813_044545  -3  0.1 1405 ~ C21TS AC6SC R-16
20250813_044600.079(0)  Transmitting 18.1 MHz + 1855Hz  FT8:  AC6SC C21TS RR73
20250813_044607.169(0)  QSO logged: AC6SC  <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
20250813_044615  -7  0.1 1405 ~ C21TS AC6SC R-16
20250813_044630.066(0)  Transmitting 18.1 MHz + 1855Hz  FT8:  AC6SC C21TS RR73
20250813_044645  -6  0.1 1405 ~ C21TS AC6SC R-16
20250813_044700.066(0)  Transmitting 18.1 MHz + 1855Hz  FT8:  AC6SC C21TS RR73
20250813_044715  -7  0.1 1405 ~ C21TS AC6SC R-16
20250813_044730.072(0)  Transmitting 18.1 MHz + 1855Hz  FT8:  AC6SC C21TS RR73
20250813_044815 -13  0.1 1405 ~ C21TS AC6SC R-16
20250813_044845 -13  0.1 1405 ~ C21TS AC6SC R-16
20250813_044900.072(0)  Transmitting 18.1 MHz + 1855Hz  FT8:  AC6SC C21TS RR73
20250813_044906.168(0)  QSO logged: AC6SC <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
```

closed 2025-08-22 by BI6NSL