foldynl/QLog #739: WinKeyer USB -  Settings

packetizeme, 2025-08-26:
### Feature Description

It would be nice if QLog supported setting the keyer mode (Iambic B, A, Ultimatic, or Bug Mode) and the ability to disable the sidetone produced by the keyer itself. Right now users will hear two sidetones unless they disable their rig sidetone.

There are many possible settings for the keyer, but these ones seem likely to vary between users more than some of the others (and are the ones that matter to me :)).

packetizeme, 2025-08-28:
Edited the are many configuration options supported by the WinKeyer, disabling sidetone is the one important to me which isn't currently present in QLog. The ability to set min and max speed for the speed pot would also be nice. I haven't used the more in-depth keyer settings and am unsure how commonly used they are, but they exist. You can see some of the options supported by the WinKeyer along with screenshots of how some other programs are presenting these in the WKUSB manual: https://www.k1elsystems.com/files/WKUSB_Manual_v1.3.pdf

packetizeme, 2025-08-28:
Thank you for letting me know about the min & max value for speed; I wasn't aware that was happening, and that behavior isn't what I expected but works for me. I don't know if this is documented anywhere and I've missed it?

packetizeme, 2025-08-29:
It doesn't look like there's a way for me to contribute to the wiki, so I'll throw this suggested patch here I suppose:
```diff --git a/Initial-Setup.md b/Initial-Setup.md
index 97d142b..9b85b5d 100644
--- a/Initial-Setup.md
+++ b/Initial-Setup.md
@@ -44,7 +44,7 @@ The CW Keyers Profile contains parameters to define a connected Keyer and its ty
 
 
 * **Keyer Mode** - currently supported key modes are: Single Paddle, Iambic A/B, Ultimate. The Key Mode is reflected only in case of Winkey model. Not all keyers support the Key Mode change
-* **Default Speed** - the speed that is set when keyer is connected. Not all keyers support the Key Speed change
+* **Default Speed** - the speed that is set when keyer is connected. If the keyer provides a speed pot for speed selection, the available speeds will be the default ± 15wpm. Not all keyers support the Key Speed change
 * **Port** - Port definition depends on platform. For Windows, there is expected string in format COMx, where x is the number of the COM port (ex. COM1, COM2 etc.). For Linux, the full path to device must be defined (ex. /dev/ttyUSB0)
 * **Baudrate** – supported values are 1200, 2400, 4800, 9600, 19200, 38400, 57600, 115200. Winkey default baudrate is set to 1200 but it can be redefined.
 * **Host Name** – IP Address or hostname where network keyer is running. Currently only CWDaemon and FLDigi are network-attachable keyers.```

foldynl, 2025-08-31:
Updated. Thanks