foldynl/QLog #750: The country or region search function is not available.

BI6NSL, 2025-09-13:
### What happened? How can it be reproduced? Provide as much information as possible.

The country or region search function is not available.

MODE search is normal

[image]

[image]

[image]

[image]

[image]

[image]

### QLog Version

0.0.0

### Which OS are you using?

Linux

### QLog Installation Method

DEB/RPM

foldynl, 2025-09-14:
Sorry, but from your description it’s not clear what you expect. Repeating the subject is not really a description of the problem.

I guess that it’s not possible to search for a country when you enter a single letter. Let me tell you that the search needs to be done in the selected language. I assume that in Chinese there is no country containing “c”, “r”, or “j”.

BI6NSL, 2025-09-14:
> I guess that it’s not possible to search for a country when you enter a single letter. Let me tell you that the search needs to be done in the selected language. 

**It is true that the corresponding country or region can be displayed in the local language, but only uppercase letters and lowercase letters and numbers can be entered in the input box, and the local language cannot be entered. Local languages can only be copied and pasted from other places.**

When you open the All Countries input interface, the input method will be forced to switch to English, and it cannot be switched.

[image]

foldynl, 2025-09-21:
Unfortunately, I don't speak Chinese and Google Translate won't translate this page. I would appreciate a  Pull Request rather than forum links. I really have no idea about keyboard input methods in Chine.