foldynl/QLog #767: Spaces after QRZ.com name

dl2ki, 2025-09-28:
### What happened? How can it be reproduced? Provide as much information as possible.

I noticed that when using a specific call sign, the name displayed in the NewQSO window was “Bernd \&nbsp;”.

[image]

A check of the source code of the QRZ.com page revealed that the name is followed by two spaces.

[image]

Perhaps the QRZ.com field contents should be cleaned up before further processing.

### QLog Version

0.0.0

### Which OS are you using?

Linux

### QLog Installation Method

DEB/RPM

foldynl, 2025-09-29:
Please, could you share a callsign? Just for testing. you can send it to me on a private email if you want.

dl2ki, 2025-10-01:
Email is on its way